ghdl/ghdl #2915: GHDL 6.0.0-dev crashes

Topi-ab, 2025-03-05:
**Description**
Bug in GHDL 6.0.0, which wasn't there on 4.x

**Expected behaviour**
Not crashing

**How to reproduce?**

```vhd :file: ghdl_bug_2025_03_05.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity ghdl_bug_2025_03_05 is
	port(
		clk_in: in std_logic;
		a_in: in std_logic;
		b_in: in std_logic
	);
end;

architecture test of ghdl_bug_2025_03_05 is
begin
	default clock is rising_edge(clk_in);
	
	
	assume always {b_in = '1'} |=> {stable(a_in)};
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 ent.vhd
```

**Context**
Please, provide the following information:

- OS:
- Origin:
https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2025-03-05/oss-cad-suite-linux-x64-20250305.tgz

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
translate_expression: cannot handle IIR_KIND_PSL_STABLE (ghdl_bug_2025-03-05.vhdl:17:41)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 6.0.0-dev (4.1.0.r593.gdd13f1d38.dirty) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/mnt/c/work/formal/fifo_assumed/
Command line:
/tools/yosys/oss-cad-suite/libexec/ghdl1-llvm --std=08 -P/tools/yosys/oss-cad-suite/lib/ghdl/ieee/v08/ -P/tools/yosys/oss-cad-suite/lib/ghdl/ -fpic -c -o ghdl_bug_2025-03-05.o ghdl_bug_2025-03-05.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x7f697bf03582 0x7f697c1027cd 0x7f697c160d58 0x7f697c160ef4 0x7f697c1623f3 0x7f697c164f7f 0x7f697c11ac18 0x7f697c10ca12 0x7f697c17f36d 0x7f697c180dbf 0x7f697c182413 0x7f697bd2b0c1 0x7f6974b77d8e 0x7f6974b77e3e 0x7f697bd298e3 0xfffffffffffffffe
******************************************************************
/tools/yosys/oss-cad-suite/libexec/ghdl:error: compilation error
```

**Additional context**

tgingold, 2025-03-05:
I am pretty sure this was not supported by ghdl 4.0.0 (and it is still not supported)

Topi-ab, 2025-03-06:
Sorry, I misread the version (v4.1.0-138...). 
Correct one seems to be 5.0.0-dev

And it synthesizes the example.

I've been using stable() criteria for a long time (which seems to be the offending part).

```
docker run --rm -it -u $(id -u):$(id -g) -v $PWD:$PWD -w $PWD hdlc/formal:all bash -c "ghdl --version; cat ghdl_bug_2025-03-05.vhdl; ghdl -a --std=08 ghdl_bug_2025-03-05.vhdl; ghdl synth --std=08 ghdl_bug_2025_03_05"
GHDL 5.0.0-dev (v4.1.0-138-ga00b7833f) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
library ieee;
use ieee.std_logic_1164.all;

entity ghdl_bug_2025_03_05 is
        port(
                clk_in: in std_logic;
                a_in: in std_logic;
                b_in: in std_logic
        );
end;

architecture test of ghdl_bug_2025_03_05 is
begin
        default clock is rising_edge(clk_in);

        assume always {b_in = '1'} |=> {stable(a_in)};
end;
library ieee;
use ieee.std_logic_1164.all;
entity ghdl_bug_2025_03_05 is
  port (
    clk_in: in std_logic;
    a_in: in std_logic;
    b_in: in std_logic
  );
end;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of ghdl_bug_2025_03_05 is
  signal wrap_clk_in: std_logic;
  signal wrap_a_in: std_logic;
  signal wrap_b_in: std_logic;
  signal n2_q : std_logic_vector (2 downto 0) := "001";
  signal n5_o : std_logic;
  signal n7_q : std_logic;
  signal n8_o : std_logic;
  signal n9_o : std_logic;
  signal n10_o : std_logic;
  signal n12_o : std_logic_vector (2 downto 0);
  signal n13_o : std_logic;
  signal n14_o : std_logic;
begin
  wrap_clk_in <= clk_in;
  wrap_a_in <= a_in;
  wrap_b_in <= b_in;
  -- ghdl_bug_2025-03-05.vhdl:17:9
  process (wrap_clk_in)
  begin
    if rising_edge (wrap_clk_in) then
      n2_q <= n12_o;
    end if;
  end process;
  -- ghdl_bug_2025-03-05.vhdl:17:9
  n5_o <= n2_q (1);
  -- ghdl_bug_2025-03-05.vhdl:17:41
  process (wrap_clk_in)
  begin
    if rising_edge (wrap_clk_in) then
      n7_q <= wrap_a_in;
    end if;
  end process;
  -- ghdl_bug_2025-03-05.vhdl:17:41
  n8_o <= '1' when wrap_a_in = n7_q else '0';
  -- ghdl_bug_2025-03-05.vhdl:17:41
  n9_o <= not n8_o;
  -- ghdl_bug_2025-03-05.vhdl:17:9
  n10_o <= n5_o and n9_o;
  n12_o <= n10_o & wrap_b_in & '1';
  -- ghdl_bug_2025-03-05.vhdl:17:9
  n13_o <= n12_o (2);
  -- ghdl_bug_2025-03-05.vhdl:17:9
  n14_o <= not n13_o;
  -- ghdl_bug_2025-03-05.vhdl:17:9
  n15: assert n14_o = '1' severity warning; --  assume
end rtl;
```

Topi-ab, 2025-03-06:
Or could it be "llvm 14.0.0 code generator" vs "static elaboration, mcode JIT code generator"?

Topi-ab, 2025-03-06:
My bad !

Changing the ghdl invocation to:

`ghdl --synth --std=08 ghdl_bug_2025-03-05.vhdl -e ghdl_bug_2025_03_05`

solved the problem. I don't need to simulate the code, only run in formal (i.e. synthesize).

Paebbels, 2025-03-06:
Your GHDL is from 28.07.2024 (a00b7833f). It's a v5.0.0-dev (138 commits after v4.1.0 was release). You could upgrade to v5.0.1 or compile from sources to get v6.0.0-dev.

Topi-ab, 2025-03-07:
> Your GHDL is from 28.07.2024 ([a00b783](https://github.com/ghdl/ghdl/commit/a00b7833f1e07c12013346a2f703810fb6c87f59)). It's a v5.0.0-dev (138 commits after v4.1.0 was release). You could upgrade to v5.0.1 or compile from sources to get v6.0.0-dev.

This is the version of last known build which works with "ghdl -a module.vhdl; ghdl synth module" way.

The version I reported (which doesn't work when separately analyzing and synthesizing) was: 4.1.0.r593.gdd13f1d38.dirty