ghdl/ghdl #2916: vhdl-2008 -

wm2015email, 2025-03-05:
Description
When compiling the design with GHDL, an error is raised regarding the constrained array slice on the port declaration. The error message

php
Copy
constrained array cannot be re-constrained
indicates that the slice notation used in the port declaration

vhd
Copy
src : in  cfixed(re to im)(left downto right);
is not acceptable to GHDL. In contrast, the same code compiles and simulates successfully with xsim.

Expected behaviour
Both simulation tools (GHDL and xsim) should accept the VHDL-2008 code if it conforms to the standard. Specifically, either:

GHDL should support the re-constraining of the array using the provided slice, or
The code should be modified to avoid re-constraining a constrained array in a way that GHDL rejects.
It is also expected that if only one component (e.g., the real part) is to be converted, the documentation and interface should clearly indicate this.

How to reproduce?

Save the following packages and entities as files:

vhd
Copy
--=================================================================================
-- cfixed_pkg
--=================================================================================
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_complex.all;
use ieee.fixed_pkg.all;
use ieee.fixed_float_types.all;

package cfixed_pkg is

  type cfixed_component_t is (re, im);
  type cfixed             is array (cfixed_component_t) of sfixed;
  
  type cfixed_vector      is array (natural range <>) of cfixed;
  
end package;
vhd
Copy
------------------------------------------------------------
-- Conversion: cfixed to float32
------------------------------------------------------------
library ieee;
use ieee.numeric_std.all;
use ieee.fixed_pkg.all;
use ieee.float_pkg.all;
use ieee.std_logic_1164.all;
use ieee.fixed_float_types.all;

use work.cfixed_pkg.all;

entity box1_cfixed_to_float32 is
    generic(
        left  : integer := 15;
        right : integer := 0
    );
    port(
        src : in  cfixed(re to im)(left downto right);
        dst : out float32
    );
end entity;

architecture hdl of box1_cfixed_to_float32 is
    
    function ele_cfixed_to_float32(val : cfixed; left : integer; right : integer) return float32 is
        variable temp : float32;
    begin
        -- Only the real part is converted
        temp := to_float(to_real(val(re)), 8, 23);
        return temp;
    end function;
    
begin
    dst <= ele_cfixed_to_float32(src, left, right);
end architecture;
Using GHDL:
Run the following command:

sh
Copy
ghdl.exe -a --std=08 -frelaxed-rules -fsynopsys -fexplicit --warn-no-hide --no-vital-checks -Wno-shared --work=work decl_usertypes_pkg.vhd try.vhd
Observed error:

vbnet
Copy
try.vhd:20:24:error: constrained array cannot be re-constrained
        src :in  cfixed(re to im)(left downto right);
                       ^
try.vhd:25:21:error: entity "box1_cfixed_to_float32" was not analysed
architecture hdl of box1_cfixed_to_float32 is
                    ^
Using xsim:
Compile and simulate with xsim using:

sh
Copy
eda_xsim.pl decl_usertypes_pkg.vhd try.vhd
Observed behavior:
The analysis logs show successful analysis and simulation:

Files analyzed without error.
Entity box1_cfixed_to_float32 is correctly processed.
Simulation completed successfully (despite a warning about missing the -top flag).
Context

OS: Tested on Windows (with paths indicating a Windows environment and OneDrive usage)
Origin:
 Package manager: version
 Released binaries: tarball_url
 Built from sources: (currently using the latest master branch)
GHDL Bug Block:
No GHDL Bug occurred block is shown in the log.
Additional context
The error seems to stem from GHDL's treatment of re-constraining a constrained array type in the port declaration. While xsim (and likely other simulators) accepts the slice notation for the cfixed type, GHDL does not allow this re-constraining under the current settings. Clarification of whether this is a GHDL limitation, a bug, or a non-conforming use of VHDL-2008 is needed. Adjusting the port declaration or modifying the type definition might be required to achieve compatibility across simulators.

Paebbels, 2025-03-05:
I fixed the formatting. Please check if it's correct.

Paebbels, 2025-03-05:
Which GHDL version was used?

Paebbels, 2025-03-05:
GHDL is correct in its behavior and if xsim accepts this code, please report a bug to AMD.

```vhdl
type cfixed_component_t is (re, im);
type cfixed             is array (cfixed_component_t) of sfixed;
```

* `cfixed_component_t` is an enumeration type, which is a scalar type and thus fully constrained.
* `cfixed` is an array type, which's index type is fully constrained and it's element type is unconstrained. Thus this type is partially constrained.

```vhdl
src: in cfixed(re to im)(left downto right);
```

The port uses a partially constrained type and constrains the missing parts. It can not re-constrain parts of the type, which are already fully constrained.

Either you:
1. define `cfixed` as unconstrained, or
   `type cfixed is array (cfixed_component_t range <>) of sfixed;`
2. skip the constraining the array index in the port  
   `src: in cfixed(open)(left downto right)`

Unike267, 2025-03-07:
To expand a bit on @diogratia's tips for Bash users:

- For single string search:

```shell
function adafind() {
  find -iname *.ad[bs] -exec grep -Hn "$1" {} \;
}
```

- For multiple strings search:

```bash
function madafind() {
  din="$1"
  shift
  for i; do
    din="${din}\|$i"
  done
  find -iname *.ad[bs] -exec grep -Hn "$din" {} \;
}
```

- Set-up steps:

1. Add the function(s) to your `.bashrc` file.
2. Reload `.bashrc` file through `. ~/.bashrc` command
3. Use the alias, i.e.:
  -  `adafind "constrained array cannot be re-constrained"`
  - `madafind "2024 Tristan" "constrained array cannot be re-constrained" "overridden"`

> [!WARNING]
> The `-Hn` options are fixed in both functions

By the way, @tgingold, I'm not a native English speaker either, but while searching for some words using these functions, I noticed that you sometimes write `overriden` instead of `overridden`. That’s a typo, right?

If so, I can prepare a PR!

tgingold, 2025-03-09:
@Unike267 Yes, you can prepare a PR.

closed 2025-03-09 by tgingold