ghdl/ghdl #2917: Simulating PSL by synthesizing the relevant parts

Topi-ab, 2025-03-08:
**Is your feature request related to a problem? Please describe.**
Several PSL features are supported on synthesis, but not on simulation.
An idea to improve on this.

**Describe the solution you'd like**
If the PSL constructs were synthesized and the synthesis netlist of those parts would be linked to rest of the VHDL file (the synthesized part is already vhdl), then all features which are supported by synth, would automatically be supported by simulation. And the same feature didn't need to be implemented twice.

**Describe alternatives you've considered**
-

**Additional context**
In addition to GHDL, I've also used Riviera Pro with PSL support. There the speed reduction due to PSL asserts can be like 50x to 100x. They probably build some kind of data structure out of PSL constructs and analyze the whole needed history on every clock cycle the assert is active.

Synthesizing and then simulating the gate level netlist should introduce almost zero overhead to simulation speed.

tgingold, 2025-03-09:
I am not sure this is the way to go forward.  During synthesis, everything becomes a net, which removes all the types.

Topi-ab, 2025-03-09:
> During synthesis, everything becomes a net, which removes all the types.

When simulating PSL constructs, are any PSL related nets shown in waveform file?

Meaning what would we lose (exactly) / be the dis-benefit, if GHDL did synthesis PSL constructs and then linking generated netlist to original VHDL file with PSL disabled?

tgingold, 2025-03-21:
I think it will be simpler to add support for prev().

tgingold, 2025-03-21:
I keep #2915 open for that.

closed 2025-03-21 by tgingold