ghdl/ghdl #2920: Winget v5.0.1 packages

Xcodo, 2025-03-10:
I would like to add the 5.0.1 packages to winget. Following comments on #2840, I suggest the following package:

`ghdl.ghdl.ucrt64.mcode`
(the format is `<publisher>.<package>.<variant>...`)

It sounds like the MinGW64 version isn't preferred, so I won't add it. Currently the release does not include other backends, so this would be a single package, replacing the current `ghdl.ghdl.ucrt64` package I asked to be created previously.

Xcodo, 2025-03-10:
@Paebbels does this sound like what you intended?

Paebbels, 2025-03-11:
Yes, this sounds good.

Thanks for explaining the naming scheme of `<publisher>.<package>....`.

Paebbels, 2025-03-15:
Is there anything to do from GHDL side?

Xcodo, 2025-03-15:
No, all set. I will try and get it done by the end of the weekend.

closed 2025-03-15 by Paebbels