ghdl/ghdl #2923: Multidimensional array assignment failure

RicardoLera, 2025-03-11:
**Description**
Attempting to assign values to a 4D array using concurrent processes results in failure.

**How to reproduce?**
The following is a minimal working example in VHDL-2008.

```vhd :file: debug.vhd
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

  entity debug is
    generic (
      n_points   : natural range 0 to 64
    );
    port (
      clock : in  std_logic
    );
  end entity debug;

architecture sim of debug is

  type b25_complex is array (0 to 1) of std_logic_vector(24 downto 0);
  type b25_complex_array is array (natural range <>) of b25_complex;
  type b25_2d_complex_array is array (natural range <>) of b25_complex_array;

  signal bfly_in : b25_2d_complex_array(0 to n_points/2-1)(0 to 1);

begin

  -- Concurrent Processes
  gen_procs_bfly : for n in 0 to n_points-1 generate

    proc_bfly : process (clock) is
      variable b, tb : integer;
    begin
      if (rising_edge(clock)) then
        report "process trigger";

        b := n/2;
        tb := n mod 2;
        
        report "n = " & integer'image(n) & "   b = " & integer'image(b) & "   tb = " & integer'image(tb) ;

        bfly_in(b)(tb) <= ("0000000010000000000000000", 25b"0");

        report integer'image(to_integer(unsigned(bfly_in(b)(tb)(0))));

      end if;
      
    end process proc_bfly;
  end generate gen_procs_bfly;

end architecture sim;
```
The purpose of this `generate` block is to describe an arbitrary number of assignments to our 4D vector `bfly_in`. We must be careful not to assign two elements in different processes, which could lead to `X` metavalues. In this configuration, that would be impossible.

The following is a simple testbench used for this example. I set a value of `n_points=2` meaning only 2 concurrent processes, which is enough for the problem to occur.

```vhd :file: debug_tb.vhd
library ieee;
  use ieee.std_logic_1164.all;
  use std.env.stop;

entity debug_tb is
end entity debug_tb;

architecture sim of debug_tb is

  component debug is
    generic (
      n_points   : natural range 0 to 64
    );
    port (
      clock   : in  std_logic
    );
  end component debug;

  signal   clock, simulate : std_logic := '0';
  constant clockperiod : time := 1 ms;

begin

  clock <= (not clock) and simulate after clockperiod / 2;

  dut : component debug
    generic map (2)
    port map (
      clock => clock
    );

  test : process is
  begin
    simulate <= '1';
    
    wait for 5 ms ;

    simulate <= '0';
    stop;
  end process test;

end architecture sim;
```
The code was compiled using:
```sh :image: ghdl/ghdl:bullseye-mcode
comp_files="debug_mwe/debug.vhd debug_mwe/debug_tb.vhd"
top_module="debug_tb"
ghdl -c --std=08 -v $comp_files -r $top_module --wave=waveforms/"${top_module%% *}".ghw --ieee-asserts=disable-at-0
```
The following is my terminal's output:
```sh :image: ghdl/ghdl:bullseye-mcode
Linking in memory
Linking in memory
Starting simulation
debug_mwe/debug.vhd:33:9:@500us:(report note): process trigger
debug_mwe/debug.vhd:38:9:@500us:(report note): n = 1   b = 0   tb = 1
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@500us:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
debug_mwe/debug.vhd:42:9:@500us:(report note): 0
debug_mwe/debug.vhd:33:9:@500us:(report note): process trigger
debug_mwe/debug.vhd:38:9:@500us:(report note): n = 0   b = 0   tb = 0
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@500us:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
debug_mwe/debug.vhd:42:9:@500us:(report note): 0
debug_mwe/debug.vhd:33:9:@1500us:(report note): process trigger
debug_mwe/debug.vhd:38:9:@1500us:(report note): n = 1   b = 0   tb = 1
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@1500us:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
debug_mwe/debug.vhd:42:9:@1500us:(report note): 0
debug_mwe/debug.vhd:33:9:@1500us:(report note): process trigger
debug_mwe/debug.vhd:38:9:@1500us:(report note): n = 0   b = 0   tb = 0
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@1500us:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
debug_mwe/debug.vhd:42:9:@1500us:(report note): 0
debug_mwe/debug.vhd:33:9:@2500us:(report note): process trigger
debug_mwe/debug.vhd:38:9:@2500us:(report note): n = 1   b = 0   tb = 1
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@2500us:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
debug_mwe/debug.vhd:42:9:@2500us:(report note): 0
debug_mwe/debug.vhd:33:9:@2500us:(report note): process trigger
debug_mwe/debug.vhd:38:9:@2500us:(report note): n = 0   b = 0   tb = 0
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@2500us:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
debug_mwe/debug.vhd:42:9:@2500us:(report note): 0
debug_mwe/debug.vhd:33:9:@3500us:(report note): process trigger
debug_mwe/debug.vhd:38:9:@3500us:(report note): n = 1   b = 0   tb = 1
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@3500us:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
debug_mwe/debug.vhd:42:9:@3500us:(report note): 0
debug_mwe/debug.vhd:33:9:@3500us:(report note): process trigger
debug_mwe/debug.vhd:38:9:@3500us:(report note): n = 0   b = 0   tb = 0
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@3500us:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
debug_mwe/debug.vhd:42:9:@3500us:(report note): 0
debug_mwe/debug.vhd:33:9:@4500us:(report note): process trigger
debug_mwe/debug.vhd:38:9:@4500us:(report note): n = 1   b = 0   tb = 1
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@4500us:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
debug_mwe/debug.vhd:42:9:@4500us:(report note): 0
debug_mwe/debug.vhd:33:9:@4500us:(report note): process trigger
debug_mwe/debug.vhd:38:9:@4500us:(report note): n = 0   b = 0   tb = 0
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@4500us:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
debug_mwe/debug.vhd:42:9:@4500us:(report note): 0
simulation stopped @5ms
```
And the waveform in gtkwave:
[image]

As evidenced, despite the reports stating that each process was triggered and the correct index values for each assignment were produced, `bfly_in` was not updated and remained undefined for all elements.

** Context**
- OS: Ubuntu 22.04.5 LTS
- Origin:
  - [X] Package manager: `4.0.0-dev`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:
[Not applicable, no bug was reported]

**Additional context**
Developed in vscodium using vhdl-ls as a linting tool, which also did not identify any bugs.

Unike267, 2025-03-12:
Hi Ricardo!

The issue is that you are injecting the value of `bfly_in(n)(b)(tb)` from `n_points-1` different sources (in this case, two). As a result, you are filling it with 'U' (unknown) values.

Imagine that each time the generate statement is applied, it creates a different concurrent module, and each of these modules assigns a value to the `bfly_in` signal simultaneously.

To solve this issue, you should define another dimension for each generate statement and initialize the signal:

```diff
  type b25_complex is array (0 to 1) of std_logic_vector(24 downto 0);
  type b25_complex_array is array (natural range <>) of b25_complex;
  type b25_2d_complex_array is array (natural range <>) of b25_complex_array;
+ type b25_3d_complex_array is array (natural range <>) of b25_2d_complex_array;

- signal bfly_in : b25_2d_complex_array(0 to n_points/2-1)(0 to 1);
+ signal bfly_in : b25_3d_complex_array(0 to n_points-1)(0 to n_points/2-1)(0 to 1):= (others => (others => (others => (others => (others => '0')))));
```

Then, adjust the signal assignment within each generate block:

```diff
-        bfly_in(b)(tb) <= ("0000000010000000000000000", 25b"0");
+       bfly_in(n)(b)(tb) <= ("0000000010000000000000000", 25b"0");

-        report integer'image(to_integer(unsigned(bfly_in(b)(tb)(0))));
+       report integer'image(to_integer(unsigned(bfly_in(n)(b)(tb)(0))));
```

With these changes, I obtained the following output from your testbench:

```
debug_or.vhd:32:9:@500us:(report note): process trigger
debug_or.vhd:37:9:@500us:(report note): n = 1   b = 0   tb = 1
debug_or.vhd:41:9:@500us:(report note): 0
debug_or.vhd:32:9:@500us:(report note): process trigger
debug_or.vhd:37:9:@500us:(report note): n = 0   b = 0   tb = 0
debug_or.vhd:41:9:@500us:(report note): 0
debug_or.vhd:32:9:@1500us:(report note): process trigger
debug_or.vhd:37:9:@1500us:(report note): n = 1   b = 0   tb = 1
debug_or.vhd:41:9:@1500us:(report note): 65536
debug_or.vhd:32:9:@1500us:(report note): process trigger
debug_or.vhd:37:9:@1500us:(report note): n = 0   b = 0   tb = 0
debug_or.vhd:41:9:@1500us:(report note): 65536
debug_or.vhd:32:9:@2500us:(report note): process trigger
debug_or.vhd:37:9:@2500us:(report note): n = 1   b = 0   tb = 1
debug_or.vhd:41:9:@2500us:(report note): 65536
debug_or.vhd:32:9:@2500us:(report note): process trigger
debug_or.vhd:37:9:@2500us:(report note): n = 0   b = 0   tb = 0
debug_or.vhd:41:9:@2500us:(report note): 65536
debug_or.vhd:32:9:@3500us:(report note): process trigger
debug_or.vhd:37:9:@3500us:(report note): n = 1   b = 0   tb = 1
debug_or.vhd:41:9:@3500us:(report note): 65536
debug_or.vhd:32:9:@3500us:(report note): process trigger
debug_or.vhd:37:9:@3500us:(report note): n = 0   b = 0   tb = 0
debug_or.vhd:41:9:@3500us:(report note): 65536
debug_or.vhd:32:9:@4500us:(report note): process trigger
debug_or.vhd:37:9:@4500us:(report note): n = 1   b = 0   tb = 1
debug_or.vhd:41:9:@4500us:(report note): 65536
debug_or.vhd:32:9:@4500us:(report note): process trigger
debug_or.vhd:37:9:@4500us:(report note): n = 0   b = 0   tb = 0
debug_or.vhd:41:9:@4500us:(report note): 65536
simulation stopped @5ms
```

I hope to be of help!

Cheers!

RicardoLera, 2025-03-13:
Thank you @Unike267 for the response! And I apologize for the slow answer.

Your workaround seems to solve the issue, despite the report not being correct at t=500, that is probably just a timing quirk of the report statement, as the waveform reports the correct result. I am, however, reluctant to add an entire new dimension of complexity to this array, when there is no reason why it should need it.

I understand that `bfly_in` is being accessed by multiple generate statements simultaneously, but the logic of the implementation **ensures** that no element of the array will be accessed by more than 1 module simultaneously. Even then, if the problem is the simultaneous access of `bfly_in` by concurrent modules like you said, how does adding one more dimension fix that issue? Is this a limitation of the VHDL-2008 standard? I simply don't see how this would be an issue in hardware

Unike267, 2025-03-13:
I can confirm that the signal `bfly_in(b)(tb)(0)` is receiving `unknown` values because I printed them using the VUnit info() function.

Unike267, 2025-03-13:
According to what I read in section `5.3.2.2`, I feel that the issue arises when you use the wrong type to refer to the index of the array. I thought it might come from the `mod` operation result, but according to `9.2.7 Multiplying Operators`, the result type of this operator is the same integer type as the first operand, in this case `n`. And `n` is a natural, right?

Tomorrow I have a meeting, and I should prepare for it, but I think I'm on the right track.

RicardoLera, 2025-03-13:
I got access to Xilinx Vivado 2023.2 and I ran a simulation using the 2008 standard. It couldn't print the waveform of the multidim arrays, but the reported values were all zero, indicating the problem is still there. This might not be an issue related to GHDL after all, my apologies.

Regarding what you found in the standard, I hadn't thought about a type mismatch but it makes sense. I'll take a look at sections `5.3.2.2` and `9.2.7` and maybe try casting some of the variables and see what results I get.

Thank you for all the help!

Unike267, 2025-03-13:
Hi again Ricardo!

Check this out:

```diff
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

  entity debug is
    generic (
      n_points   : natural range 0 to 64
    );
    port (
      clock : in  std_logic
    );
  end entity debug;

architecture sim of debug is

  type b25_complex is array (0 to 1) of std_logic_vector(24 downto 0);
  type b25_complex_array is array (natural range <>) of b25_complex;
  type b25_2d_complex_array is array (natural range <>) of b25_complex_array;

- signal bfly_in : b25_2d_complex_array(0 to n_points/2-1)(0 to 1);

begin

  -- Concurrent Processes
  gen_procs_bfly : for n in 0 to n_points-1 generate

    proc_bfly : process (clock) is
      variable b, tb : integer;
+     variable bfly_in : b25_2d_complex_array(0 to n_points/2-1)(0 to 1) := (others => (others => (others => (others => '0'))));
    begin
      if (rising_edge(clock)) then
        report "process trigger";

        b := n/2;
        tb := n mod 2;
        
        report "n = " & integer'image(n) & "   b = " & integer'image(b) & "   tb = " & integer'image(tb) ;

-        bfly_in(b)(tb) <= ("0000000010000000000000000", 25b"0");
+        bfly_in(b)(tb) := ("0000000010000000000000000", 25b"0");

        report integer'image(to_integer(unsigned(bfly_in(b)(tb)(0))));

      end if;
      
    end process proc_bfly;
  end generate gen_procs_bfly;

end architecture sim;
```

RicardoLera, 2025-03-17:
@diogratia Thank you as well for the answer and I apologize for the late response.

Regarding the term "multidimensional", if you read the comments so far you'd know that I acknowledged this difference and retested this problem using actual multidimensional signals. The name of the Issue is valid.

Regarding the explanation, allow me to paraphrase the standard:
```
Given
    S   - scalar signal [bfly_in(b)(tb)]
    LSP - Longest Static Prefix [bfly_in]
    SAS - Signal Assignment Statement [bfly_in(b)(tb) <= ("0000000010000000000000000", 25b"0");]

Then, in EACH process
    IF   There is at least 1 SAS that targets S [TRUE]
    AND  The LSP of S denotes S or a composite signal of which S is a subelement [TRUE]
    THEN There is 1 driver for S, to which this SAS is associated exclusively
```
Final result: There are `n_points` drivers for `bfly_in(b)(tb)`, to which each respective SAS is associated to exclusively.
And the indexes `b` and `tb` never clash.
Therefore at **no point** do two drivers attempt to drive the same static signal.
So how exactly does this cause a conflict?

Your logic explains @Unike267's second workaround (if `bfly_in` is not static, the logic fails, making no drivers) but does not explain why the indexes being constants would change anything besides the LSP, which is only used to check for the association between S and SAS.

Unless you mean to say that each of these drivers is **not** associated with the scalar signal S described in the SAS (`bfly_in(t)(tb)`) but instead associated with the *composite signal denoted by the longest static prefix* (`bfly_in`) thus causing a clash. A behavior that **contradicts** Section 14.7.2 Drivers:

> There is a single driver for a given scalar signal S in a process statement.

diogratia, 2025-03-18:
>Final result: There are n_points drivers for bfly_in(b)(tb), to which each respective SAS is associated to exclusively.
And the indexes b and tb never clash.  
Therefore at **no point** do two drivers attempt to drive the same static signal.
So how exactly does this cause a conflict?

Your statement preceding the question in the above excerpt from your comment is incorrect.

See the 8. Names, 8.1 General quote in my first comment. I didn't quote a paragraph from 8.1 that would have been helpful here:

>A name is said to be a _static name_ if and only if one of the following conditions holds:  
>>— The name is a simple name or selected name (including those that are expanded names) that does not denote a function call, an object or value of an access type, or an object of a protected type and (in the case of a selected name) whose prefix is a static name.  
>>— The name is an indexed name whose prefix is a static name, and every expression that appears as part of the name is a static expression.

The expression here is found in the index values b and tb.  Also while terms of art specific to VHDL are defined in the semantic text of the standard, they can also be found along with links to specific subsections of the standard in the standard's glossary which also provides a synopsys of definitions.

Variables b and tb are not static expressions.  (Neither are signals, but that's a non sequitur).

9.4 Static expressions  
9.4.1 General  

Certain expressions are said to be _static_. Similarly, certain discrete ranges are said to be static, and the type
marks of certain subtypes are said to denote static subtypes.  
>  
>There are two categories of static expression. Certain forms of expression can be evaluated during the analysis of the design unit in which they appear; such an expression is said to be _locally static_. Certain forms of expression can be evaluated as soon as the design hierarchy in which they appear is elaborated; such an expression is said to be _globally static_.

Variables don't show up in the table in 9.4.2 Locally static primaries nor in the table found in 9.4.3 Globally static primaries. The longest static prefix of `bfly_in(b)(tb)` is `bfly_in`. There are drivers for each scalar subelement of bfly_in (which is a signal) for each process statement in the generate statement elaborated block statements (11.8 Generate statements, 14.5.3 Generate statements). Where the value of of scalar subelements of bfly_in are determined by a resolution function.

An indexed name isn't static if the index has non-static expressions. If instead of b and tb you used numerical literals falling within index ranges for each type you'd have a longest static prefix containing the indexes. Constants are also static expressions (See 9.4), explaining why declaring b and tb as constants give a longest static prefix resolving to a subelement and working as you intended. 

Another way besides declaring b and tb as constants to demonstrate you have multiple drivers is to switch from an element type of std_logic to bit (which here would require switching from std_logic_vector to bit_vector and package numeric_std to numeric_bit for the conversion function to_integer as well as the unsigned declaration). 6.4.2.3 Signal declarations " is an error if, after the elaboration of a description, a signal has multiple sources and it is not a resolved signal." Type bit is not resolved. 

```vhdl
library ieee;
  use ieee.std_logic_1164.all;
  -- use ieee.numeric_std.all;
  use ieee.numeric_bit.all;

  entity debug is
    generic (
      n_points   : natural range 0 to 64
    );
    port (
      clock : in  std_logic
    );
  end entity debug;

architecture sim of debug is

  -- type b25_complex is array (0 to 1) of std_logic_vector(24 downto 0);
  type b25_complex is array (0 to 1) of bit_vector(24 downto 0);
  type b25_complex_array is array (natural range <>) of b25_complex;
  type b25_2d_complex_array is array (natural range <>) of b25_complex_array;

  signal bfly_in : b25_2d_complex_array(0 to n_points/2-1)(0 to 1);

begin
```
The run time error message is distinct (here for an llvm code converter version of ghdl):

```none
%: ghdl -a --std=08 debug_tb.vhdl
%: ghdl -e --std=08 debug_tb
%: debug_tb --stop-time=2us
Signal .debug_tb(sim).dut@debug(sim).bfly_in(0)(0)(0)(24) is driven by :
.debug_tb(sim).dut@debug(sim).gen_procs_bfly(0).proc_bfly
.debug_tb(sim).dut@debug(sim).gen_procs_bfly(1).proc_bfly
debug_tb:error: several sources for unresolved signal
debug_tb:error: error during elaboration
%: 
```

For an llvm JIT or mcode version you might see a large number of less informative error messages. The difference is the llvm code converter version quits after attempting to resolve the first set of drivers on a net while specifying the processes. The llvm JIT version produces error messages for each subelement driver of signal bfly_in before quitting without specifying the processes or the indexed names, rather reporting the declaration of bfly_in.

As you might gather understanding longest static prefix along with some basic knowledge of resolution can be necessary in settings where subsets of elements of composite signals are assigned in multiple processes. That includes as here in generate statements.

The good news is once you get this stuff straight in your head you're likely to remember it.

RicardoLera, 2025-03-18:
@diogratia Thank you for taking the time to explain and source the reference manual, I spent some time reading it myself. I think I misinterpreted the paragraph from 14.7.2 Drivers

> (...) There is a single driver for a given scalar signal S in a process statement, provided that there is at least one
signal assignment statement in that process statement and that the longest static prefix of the target signal of
that signal assignment statement denotes S or denotes a composite signal of which S is a subelement. (...)

I was under the impression this meant only 1 driver could be generated per signal assignment statement per process in a generate loop. Instead, the limitation is only per *scalar signal*, per process. Each process in my code creates a driver for each subelement of `bfly_in`, due to it being the longest static prefix, causing a conflict.

Following that logic, I have found a way to fix the issue with no warnings in GHDL:
```
  gen_procs_bfly : for b in 0 to n_points/2-1 generate
    gen_procs_bfly_tb : for tb in 0 to 1 generate
      proc_bfly : process (state) is
      begin
        for n in 0 to n_points-1 loop
          if ((b = bfly_lut(state, n)(0)) and (tb = bfly_lut(state, n)(1))) then
            bfly_in(b)(tb) <= (std_logic_vector(to_unsigned(state, 25)), 25b"0");
          end if;
        end loop;
      end process proc_bfly;
    end generate gen_procs_bfly_tb;
  end generate gen_procs_bfly;
```
This snippet is a bit different from the previous examples (I applied it to my main project), but it exemplifies the use with a state machine, meaning different values can be assigned at different clock times. As you can see, `b` and `tb` are now generate parameters, which are static. This allows me to iterate along the array on each state and assign whichever values I want while only generating drivers to the respective elements.

Thank you again, to the both of you, for having patience in discussing this issue with me. I will be closing the issue with this comment.

closed 2025-03-18 by RicardoLera