ghdl/ghdl #2924: Codacy Static Code Analysis Still Running After 25 Days

Unike267, 2025-03-13:
Hi there!

The following Codacy Static Code Analysis job has been running for 25 days. Is this normal behavior?

https://github.com/ghdl/ghdl/runs/37281272209

Cheers!

tgingold, 2025-03-13:
No, but we don't care!  Probably codacy was not able to send back the results.
(I am not even sure it is still enabled).

closed 2025-03-13 by tgingold

Unike267, 2025-03-13:
I phoned Unai to tell him about the issue before opening it, to see if he could fix it, and he told me that it seems you modified the Codacy Static Code Analysis a bit. Unfortunately, he couldn’t stop it from his account.

Thanks, Tristan!

Paebbels, 2025-03-13:
I can see GHDL in my Codacy account, but I can't modify it. It reports I have no permissions.