ghdl/ghdl #2925: Checking if the following errors are because limitation of GHDL

jefflieu, 2025-03-15:
Hi,
I've been using Aldec simulator and 
I'm exploring the speed potential of GHDL.
However, I'm running into some problem compiling existing working VHDL2008 code. 
I just want to check if these are genuine limitation of GHDL and if these issues are being worked on.
-------------
I've attached 3 files and compiling them leads me to the following error. 

```
test_ghdl.vhd:17:36:error: 'others' choice not allowed for an aggregate in this context
  unconstrained_record.signal_a <= (others => '0');
```
And 
```
test_ghdl_top.vhd:29:39:error: type of actual conversion must be fully constrained
    unconstrained => std_ulogic_vector(constrained),
```

jefflieu, 2025-03-15:
[test_ghdl.zip](https://github.com/user-attachments/files/19257357/test_ghdl.zip)

diogratia, 2025-03-15:
ghdl versions used:

```none
%: ghdl --version
GHDL 6.0.0-dev (b6.0.0.r7.g030fe0890) [Dunoon edition]
 Compiled with GNAT Version: 14.2.0
 static elaboration, LLVM JIT code generator
Written by Tristan Gingold.

```

and

```
%: ghdl --version
GHDL 6.0.0-dev (b6.0.0.r7.g030fe0890) [Dunoon edition]
 Compiled with GNAT Version: 14.2.0
 llvm 19.1.7 code generator
Written by Tristan Gingold.

````
The platform is MacOS 15. Both the llvm JIT and llvm back end versions produce the same results.

These are not ghdl limitations, rather found in the semantics of the standard.

```none
test_ghdl.vhd:17:36:error: 'others' choice not allowed for an aggregate in this context
  unconstrained_record.signal_a <= (others => '0');
```

9.3.3.3 Array aggregates

>The index range of an array aggregate that has an **others** choice shall be determinable from the context. That
is, an array aggregate with an **others** choice shall appear only in one of the following contexts:
>  __
>>a) As an actual associated with a formal parameter, formal generic, or formal port (or member thereof), where either the formal (or the member) is declared to be of a fully constrained array subtype, or the formal designator is a slice name    
>>b) As the default expression defining the default initial value of a port declared to be of a fully constrained array subtype  
>>c) As the default expression for a generic constant declared to be of a fully constrained array subtype  
>>d) As the result expression of a function, where the corresponding function result type is a fully constrained array subtype  
>>e) As a value expression in an assignment statement, where the target is a declared object (or member thereof), and either the subtype of the target is a fully constrained array subtype or the target is a slice name  
>>f) As the expression defining the initial value of a constant or variable object, where that object is declared to be of a fully constrained array subtype  
>>g) As the expression defining the default values of signals in a signal declaration, where the corresponding subtype is a fully constrained array subtype  
>>h) As the expression defining the value of an attribute in an attribute specification, where that attribute is declared to be of a fully constrained array subtype  
>>i) As the operand of a qualified expression whose type mark denotes a fully constrained array subtype  
>>j) As a choice in a case statement whose expression is of a one-dimensional character array type and is one of the following:  
>>>— The name of an object whose subtype is locally static, in which case the index range of the aggregate is the index range of the subtype of the object  
>>>— An indexed name whose prefix is one of the members of this list and whose indexing expressions are locally static expressions, in which case the index range of the aggregate is the index range of the element subtype of the prefix  
>>— A slice name whose prefix is one of the members of this list and whose discrete range is a locally static discrete range, in which case the index range of the aggregate is the discrete range  
>>— A function call whose return type mark denotes a locally static subtype, in which case the index range of the aggregate is the index range of the subtype denoted by the return type mark  
>>— A qualified expression or type conversion whose type mark denotes a locally static subtype, in which case the index range of the aggregate is the index range of the subtype denoted by the type mark  
>>— An expression described in this list and enclosed in parentheses, in which case the index range of the aggregate is the index range of the subtype defined for the enclosed expression  
>  
>In each case, the applicable index constraint is locally static.  
>  
>k) As a subaggregate nested within an aggregate, where that aggregate itself appears in one of these contexts

The candidate context for the use of the **others** choice above is e)but the assignment target is not fully constrained:

5. Types
5.1 General 

>A composite subtype is said to be _unconstrained_ if:  
>>— It is an array subtype with no index constraint and the element subtype either is not a composite subtype or is an unconstrained composite type, or  
>>— It is a record subtype with at least one element of a composite subtype and each element that is of a composite subtype is unconstrained.  
>  
>A composite subtype is said to be _fully constrained_ if:  
>>— It is an array subtype with an index constraint and the element subtype either is not a composite subtype or is a fully constrained composite type, or  
>>— It is a record subtype and each element subtype either is not a composite subtype or is a fully constrained composite subtype.  
>  
>A composite subtype is said to be _partially constrained_ if it is neither unconstrained nor fully constrained.
[... 89 lines cut ...]
end behaviour;
```
it analyzes, elaborates and simulates and we find your model attempts to run until TIME'high unless stopped by command line argument. There are other possible fixes.

Belatedly finding the reference for the aggregate using find:

```none
%: adafind -n "in this context"
./vhdl/vhdl-sem_expr.adb:4087:                                         & "for an aggregate in this context");
```
where it took successive truncation to find the particular error message. Previous revisions of the standard section 7.3.2.2 Array aggregates have the same restriction on the use of **others**:

[... cut ...]
>>b) The time-variant state of all signals and variables in the description are the same at all times during
the simulation, under the condition that the same stimuli are applied at the same times to the description. The stimuli applied to a model include the values supplied to generics and ports at the root of the design hierarchy of the model, if any.

And here ghdl is claimed to be a conformant implementation for these two errors. That would imply your existing working VHDL2008 code isn't as portable as you have assumed.

jefflieu, 2025-03-16:
As I put in the title, I don't know what is correct, and the only "reference" that I have is that it works with other tools.

Thanks for taking time to post the LRM of the language. 
I didn't learn VHDL or VHDL2008 by reading LRM.
I don't think any one does that. It's learned by examples.
I guess I'm `spoilt` by some relaxation of the standard along the way without even knowing it.

closed 2025-03-16 by jefflieu

diogratia, 2025-03-16:
>I didn't learn VHDL or VHDL2008 by reading LRM.  
I don't think any one does that. It's learned by examples.

I used to liken liken progression in VHDL to the Journeyman system responsible for ending the Dark Ages in Europe and parallel in university systems. Where the Journeyman has apprentices, journeymen, and guild masters, universities have Baccalaureate, masters and Doctorate programs

Unike267, 2025-03-16:
> the Journeyman system

Thank goodness we don't wear those hats anymore! Nowadays, we’ve swapped them for Red Hat haha

---

BTW Jeff, I don’t know how many of us could give you an accurate/correct response. There are a lot of brilliant people here, but they are very busy. I think the only one who has the knowledge and the time to help with that kind of answer is Dio.