ghdl/ghdl #2928: Crash when merging coverage files

cwyki, 2025-03-16:
**Description**
The tool `ghdl coverage` crashes when merging coverage results when the corresponding simulations have common files.
I am not 100% sure if that is a bug or just not supported, but I assumed that it would be possible to merge simulations which use (partially) the same files.

The call to `ghdl coverage` is also used in vunits [ghdl.py](https://github.com/VUnit/vunit/blob/acf7e7f294f9b1b11dc664c2f145b9cb95906532/vunit/sim_if/ghdl.py#L466)  and when merging different vunit simulations there are several .vhd files from vunit itself which are used in all the simulations.

**Expected behaviour**
Merge coverage JSONs without a crash. A line of a file would be considered as covered if it is covered in at least one simulation.

**How to reproduce?**

```vhd :file: reproducer.vhd
library ieee;
use ieee.std_logic_1164.all;
entity reproducer is
    port (
        clk : in std_logic
    );
end entity reproducer;

architecture rtl of reproducer is
    SIGNAL a : std_logic;
begin
    a <= '1' when rising_edge(clk);
end architecture;
```
```vhd :file: tb_reproducer.vhd
library ieee;
use ieee.std_logic_1164.all;
entity tb_reproducer is
end entity tb_reproducer;

architecture rtl of tb_reproducer is
    signal clk : std_logic := '0';
begin
    reproducer_inst : entity work.reproducer(rtl)
        port map(
            clk => clk
        );
    main : process
    begin
        wait;
    end process;
end architecture;
```

```vhd :file: tb_second.vhd
library ieee;
use ieee.std_logic_1164.all;

entity tb_second is
end entity tb_second;

architecture rtl of tb_second is
    signal clk : std_logic := '0';
begin
    reproducer_inst : entity work.reproducer
        port map(
            clk => clk
        );

    main : process
    begin
        wait;
    end process;
end architecture;
```
```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --work=lib --std=08 reproducer.vhd tb_reproducer.vhd tb_second.vhd 
ghdl --elab-run --work=lib --std=08  --coverage --coverage-output=tb_reproducer.json tb_reproducer
ghdl --elab-run --work=lib --std=08  --coverage --coverage-output=tb_second.json tb_second
ghdl coverage --format=lcov tb_reproducer.json tb_second.json
```
The last call produces the bug shown below.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager:
  - [ ] Released binaries: 
  - [x] Built from sources: `37ad91899ea3f311423eb91e34f42c0ae1948f79` (Release 5.0.1)

Bug message:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.1 (tarball) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-linux-gnu
/var/local/reproducer/
Command line:
ghdl coverage --format=lcov tb_reproducer.json tb_second.json
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : ghdlcov.adb:328 access check failed
******************************************************************
```

**Additional context**
None

cwyki, 2025-03-18:
There is also a small issue with the output `--format=lcov`. Using the files like in the description and executing `ghdl coverage --format=lcov tb_reproducer.json` produces the output
```
TN:
SF:tb_reproducer.vhd
FN:1:file
FNDA:1,file
DA:17,1
end_of_record
SF:reproducer.vhd
FN:1:file
FNDA:1,file
DA:13,1
end_of_record
```
The line `FN:1:file` should read `FN:1,file` (comma instead of colon) according to the format description 
from [https://manpages.debian.org/testing/lcov/geninfo.1.en.html](https://manpages.debian.org/testing/lcov/geninfo.1.en.html):
```FN:<line number of function start>,[<line number of function end>,]<function name>```

tgingold, 2025-03-19:
Thank you for the report.  I am testing a fix.

commit 7df59127 (2025-03-19): ghdlcov: fix a crash, adjust lcov output

Fix #2928
closed 2025-03-19 by tgingold