ghdl/ghdl #2930: Synthesis of memory takes too much time if there is init

marzoul, 2025-03-17:
**Description**
Synthesis of memory takes too much time if there is init.
Loading design takes no time if there is no init, but it takes around 7 seconds if there is init (laptop computer).
Note that the memory size is 2^24 = 16 million lines, this is not small but not that high either, and this kind of size is getting very common.

**Expected behaviour**
Synthesis should not take a noticeably higher amount of time than without init, especially with the same init for all words such as `(others => '0')`.

**How to reproduce?**

```vhd :file: top.vhd

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity top is
	generic (
		DATA_BITS : natural := 32;
		CELLS     : natural := 2 ** 24;
		ADDR_BITS : natural := 32
	);
	port (
		clk   : in  std_logic;
		wen   : in  std_logic;
		waddr : in  std_logic_vector(ADDR_BITS-1 downto 0);
		wdata : in  std_logic_vector(DATA_BITS-1 downto 0);
		ren   : in  std_logic;
		raddr : in  std_logic_vector(ADDR_BITS-1 downto 0);
		rdata : out std_logic_vector(DATA_BITS-1 downto 0)
	);
end top;

architecture synth of top is

	type mem_type is array(0 to CELLS-1) of std_logic_vector(DATA_BITS-1 downto 0);
	--signal mem : mem_type;
	signal mem : mem_type := (others => (others => '0'));

	attribute ram_style : string;
	attribute ram_style of mem : signal is "block";

begin

	process(clk)
	begin
		if rising_edge(clk) then

			if ren = '1' then
				rdata <= mem(to_integer(unsigned(raddr)));
			end if;

			if wen = '1' then
				mem(to_integer(unsigned(waddr))) <= wdata;
			end if;

		end if;
	end process;

end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a top.vhd
ghdl --synth --out=verilog -o=/dev/null top
```

**Context**

- OS: Archlinux
- Origin: Built from sources: `commit 19281ad215eb96ab8e4c2dab6e3e141dceae99fb` done on 2025-02-26

**Additional context**

This is also affecting the GHDL plugin for Yosys, see this for reference : https://github.com/YosysHQ/yosys/issues/4601
This means that the production of millions of lines of verilog is not the source of the issue.
gdb shows that most of the time is spent within function `synth.vhdl_expr.logic2logvec`.

tgingold, 2025-03-17:
I will try to optimize this case, but why do you initialize memories ?

Paebbels, 2025-03-17:
Xilinx BlockRAM can be initialed at FPGA startup. See the attribute: he sets the style to `block`.

tgingold, 2025-03-18:
Sure, but in most/all cases, this is used for ROM, not for zero-ing a RAM.

Paebbels, 2025-03-18:
Common use cases for (Block)RAM inits are:
* Initial boot code, microcode for µP.  
  These can be updated at runtime via JTAG or UART or similar.
* Initial coefficients for calculations / DSPs.  
  These can be swapped at runtime to adjust filters.

Paebbels, 2025-03-18:
In addition for zeroing RAMs:  
Many FPGA RAMs (Xilinx, Altera, ...) are guaranteed to start at zero (if not initialized otherwise), thus this should be part of the simulation model (VHDL code). An init of `U` is not correct for these technologies.

marzoul, 2025-03-18:
If the mem init can also be improved for arbitrary data, this would be super welcome too.
In many of my designs there are matrices of parameters or bunches of arbitrary parameters, these will become ROMs in an FPGA design.

Users often have some flexibility in how the data is generated, and there is probably a format that best pleases GHDL. Is there, which one ?
Declaration as array or std_logic_vector, array of unsigned, array of bit_vector ?
Maybe the style of the initialization list matters : hex, binary as string, integer ?

Unike267, 2025-03-19:
Hi Adrien!

> Maybe the style of the initialization list matters : hex, binary as string, integer ?

You're likely familiar with this RAM/ROM initialization mode, but this might save you some work!

https://github.com/Unike267/afselerator/blob/e7f3702013cfdb3bf6682ab7e8fa91a6ef572d63/rtl/MEM/MEM_package.vhd#L36-L49

https://github.com/Unike267/afselerator/blob/e7f3702013cfdb3bf6682ab7e8fa91a6ef572d63/rtl/MEM/ROM/ROM.vhd#L39-L56

Unike267, 2025-03-19:
That's a really interesting point of view!

Actually, I haven't performed a timing test using GHDL yet. It would be interesting to take measurements for:

https://github.com/ghdl/ghdl/blob/2b9d68e6b5a87a4894668ebf59b5973bd21a0aad/libraries/ieee2008/std_logic_1164.vhdl#L283
https://github.com/ghdl/ghdl/blob/2b9d68e6b5a87a4894668ebf59b5973bd21a0aad/libraries/ieee2008/std_logic_1164.vhdl#L288
https://github.com/ghdl/ghdl/blob/2b9d68e6b5a87a4894668ebf59b5973bd21a0aad/libraries/ieee2008/std_logic_1164.vhdl#L293

to practically analyze the issue.

Unike267, 2025-03-19:
Wow, now I'm a little bit confused.

I've seen that these functions are also defined in:

https://github.com/ghdl/ghdl/blob/2b9d68e6b5a87a4894668ebf59b5973bd21a0aad/libraries/std/textio.vhdl#L97-L116

I know that some functions used in:

> https://github.com/Unike267/afselerator/blob/e7f3702013cfdb3bf6682ab7e8fa91a6ef572d63/rtl/MEM/MEM_package.vhd#L36-L49

rely on the `textio` package to execute certain procedures, such as `readline`.

However, in the case of the `HREAD` function, I don't know exactly which package it depends on.

Unike267, 2025-03-19:
> However, in the case of the HREAD function, I don't know exactly which package it depends on.

I've checked this question with the following `test.vhd`:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.env.stop;

library MEM;
use MEM.MEM_package.all;

entity test is

end test;

architecture rtl of test is

  constant clk_period      : time      := 10 ns;
  signal   clk             : std_logic := '0';
  constant ROM_DEPTH       : natural   := 3; 
  constant ROM_LOAD_FILE_8 : string    := "string.hex"; 
  constant ROM_8           : t_MEM8 (0 to ( 2**ROM_DEPTH)-1) := MEM8_INIT_HEX (ROM_LOAD_FILE_8,  2**ROM_DEPTH);

begin

  clk <= not clk after clk_period/2;

  report_value : process (clk) is
    begin
      if (rising_edge(clk)) then
         report "The sixth position (in decimal) is: " & integer'image(to_integer(unsigned(ROM_8(6))));
      end if;
    end process report_value;

  testing : process is
  begin
    wait for 10 ns;
    stop;
  end process testing;

end rtl;
```

In the same directory as `MEM_package.vhd`, `test.vhd`, `string.hex` and the build directory, run:

- `ghdl -i --std=08 --workdir=build --work=MEM *.vhd`
- `ghdl -m --std=08 --workdir=build --work=MEM test`
- `ghdl -r --std=08 --workdir=build --work=MEM test`

With this `string.hex`:

```
00
11
22
33
AA
11
22
33
```

The output is:

```
test.vhd:29:10:@5ns:(report note): The sixth position (in decimal) is: 34
simulation stopped @10ns
```

Now, changing the sixth position, which forces an error

```diff
00
11
22
33
AA
11
- 22
+ 2ñ
33

```

and re-running the test, he output is:

```
../../src/ieee2008/std_logic_1164-body.vhdl:1441:9:@0ms:(assertion error): STD_LOGIC_1164.HREAD Error: Read a '�', expected a Hex character (0-F).
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@5ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
test.vhd:29:10:@5ns:(report note): The sixth position (in decimal) is: 0
simulation stopped @10ns
```

After checking:

https://github.com/ghdl/ghdl/blob/2b9d68e6b5a87a4894668ebf59b5973bd21a0aad/libraries/ieee2008/std_logic_1164-body.vhdl#L1443-L1444

It seems that `HREAD` comes from the `std_logic_1164` package!

Unike267, 2025-03-19:
> @Unike267 I think this specific discussion should be placed in the context of the repository of interest ;-)

Sorry, it was a spontaneous doubt, take it as if I were thinking out loud.

BTW, I tried loading a `2 ** 24` 32-bit hex file (to check if it takes more than 7 seconds) using:

```diff

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

+ -- Change from MEM to MEM_MGMT (memory management) for not hide library "mem" with declaration of "mem" signal
+ library MEM_MGMT;
+ use MEM_MGMT.MEM_package.all;

entity top is
	generic (
		DATA_BITS : natural := 32;
		CELLS     : natural := 2 ** 24;
		ADDR_BITS : natural := 32
	);
	port (
		clk   : in  std_logic;
		wen   : in  std_logic;
		waddr : in  std_logic_vector(ADDR_BITS-1 downto 0);
		wdata : in  std_logic_vector(DATA_BITS-1 downto 0);
		ren   : in  std_logic;
		raddr : in  std_logic_vector(ADDR_BITS-1 downto 0);
		rdata : out std_logic_vector(DATA_BITS-1 downto 0)
	);
end top;

architecture synth of top is

-	type mem_type is array(0 to CELLS-1) of std_logic_vector(DATA_BITS-1 downto 0);
-	--signal mem : mem_type;
-	signal mem : mem_type := (others => (others => '0'));

+    constant ROM_LOAD_FILE_32 : string := "string.hex";
+    signal mem :t_MEM32 (0 to CELLS-1) := MEM32_INIT_HEX (ROM_LOAD_FILE_32,  CELLS);

	attribute ram_style : string;
	attribute ram_style of mem : signal is "block";

begin

	process(clk)
	begin
		if rising_edge(clk) then

			if ren = '1' then
				rdata <= mem(to_integer(unsigned(raddr)));
			end if;

			if wen = '1' then
				mem(to_integer(unsigned(waddr))) <= wdata;
			end if;

		end if;
	end process;

end architecture;
```

And tested it with the following testbench:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.env.stop;

library MEM_MGMT;
use MEM_MGMT.MEM_package.all;

entity test is

end test;

architecture tb of test is

  constant clk_period : time      := 10 ns;
  signal   clk        : std_logic := '0';

  signal ren   : std_logic := '0';
  signal wen   : std_logic := '0';
  signal waddr : std_logic_vector(31 downto 0) := (others => '0');
  signal wdata : std_logic_vector(31 downto 0) := (others => '0');
  signal raddr : std_logic_vector(31 downto 0) := (others => '0');
  signal rdata : std_logic_vector(31 downto 0) := (others => '0');

begin

  uut : 
  entity MEM_MGMT.top
                     generic map(
                                DATA_BITS   => 32,
                                CELLS => 2 ** 24,
                                ADDR_BITS => 32
                                )
                     port    map(
                                clk   => clk,
                                wen   => wen,
                                waddr => waddr,
                                wdata => wdata,
                                ren   => ren,
                                raddr => raddr,
                                rdata => rdata
                                );

  clk <= not clk after clk_period/2;

  read_value : process (clk) is
    begin
      if (rising_edge(clk)) then
         ren <= '1';
         raddr <= std_logic_vector(to_unsigned(8388607, 32));
      end if;
    end process read_value;

  testing : process is
  begin
    wait for 2*clk_period;
    -- For this address the output must match 0x8dc3284f (2378377295)
    report "The position eight million three hundred eighty-eight thousand six hundred eight (in decimal) is: " & integer'image(to_integer(unsigned(rdata)));
    stop;
  end process testing;

end tb;
```

With this setup, I executed:

- `ghdl -i --std=08 --workdir=build --work=MEM_MGMT *.vhd`
- `ghdl -m --std=08 --workdir=build --work=MEM_MGMT test`
- `ghdl -r --std=08 --workdir=build --work=MEM_MGMT test`

But I encountered the following error:

```
./test:error: declaration of a too large object (524288 kB > --max-stack-alloc=128 kB)
  instance: 
./test:error: error during elaboration
```

So I increased the max stack allocation using:

- `ghdl -r --std=08 --workdir=build --work=MEM_MGMT test --max-stack-alloc=524288`

However, the execution crashes without any error message from GHDL, but the system does produce an error:

[image]

The `hex` file occupies 150,994,943 bytes. I tried attaching it, but it's too large (I was able to open it in Emacs without any issues).

@tgingold, any hints?

tgingold, 2025-03-20:
I have optimized the handling of large memories.  I still have some ideas to further optimize, but at some point I would need a more realistic reproducer!

tgingold, 2025-03-20:
@Unike267 If this is a crash, please open a new issue.

tgingold, 2025-03-21:
I added a last bit of optimization.  Again, if it is still too slow, please reopen with a reproducer.

marzoul, 2025-03-24:
> I have optimized the handling of large memories. I still have some ideas to further optimize, but at some point I would need a more realistic reproducer!

Thank you ! I can see improvement indeed.
The follow-up of this topic will be in the context of the ghdl plugin for Yosys, which seems to have its own scalability issues (issues that Slang does not have to load the same design in verilg).

tgingold, 2025-03-25:
Using which testcase ?

marzoul, 2025-03-26:
The same testcase. Juste a zero-initialized memory with sync read and write so it can be mapped on usual BRAM.
What I observe is that, when such design is loaded in Yosys with ghdl plugin, it triggers much more burden on Yosys backend (netlist simplification etc) than when the equivalent design in verilog is loaded with the Slang plugin.

The topic emerged in Yosys issue https://github.com/YosysHQ/yosys/issues/4601
And verilog example in comment https://github.com/YosysHQ/yosys/issues/4601#issuecomment-2714933545
I have to re-run everything on my side, advise on what size to use for a proper reproducer, etc.

tgingold, 2025-03-26:
@marzoul, can you show the rtlil generated by slang for the zero-initialized memory ?

marzoul, 2025-03-28:
Update : After corrections and update of all tools, ghdl + slang + yosys, the run time and ram usage numbers have changed

@tgingold Here is the archive in a public cloud :
https://cloud.univ-grenoble-alpes.fr/s/ebaMW48KycZpsQQ

It is an simple 2-port memory, 32b data, 2^20 - one million lines, zero-initialized.
It gets synthesized into 1024 BRAM blocks. The burden on Yosys seems to be simplification passes on mux expressions, which it seems to be doing repeatedly on every bit of the 32b datapath, but that's another issue.

Files `ghdl-rtlil.log` and `slang-rtlil.log` are Yosys log files when loading the design with respectively ghdl and slang plugins. Something worth noting : slang takes 2x the time and 3x the RAM usage than ghdl plugin, I know there are inefficiencies on slang code side, but slang seems to launch things in Yosys after reading verilog, which may explain the difference.

However for synthesis things are different, see end of `ghdl-synth.log` and `slang-synth.log`.
- with ghdl plugin, yosys takes 32 sec and 1.8 GB of RAM
- with slang plugin, yosys takes 50 sec and 1.2 GB of RAM
Synthesis results look similar. With ghdl, run time is significantly lower ( 👍 ) but RAM usage is +500 MB (??).

I don't know if there is something to optimize here. Saving 500 MB of RAM is always appreciated. Feel free to re-close this issue if there is nothing obvious to do now.

tgingold, 2025-03-29:
For the memory size and speed, I don't think they can be significantly improved.

I haven't looked at the difference between ghdl and slang output.  If needed I could.

closed 2025-03-29 by tgingold