ghdl/ghdl #2931: Update of ghdl@5 package under guix

csantosb, 2025-03-17:
Hi, I’m trying to [update](https://codeberg.org/guix-science/guix-science/pulls/87) the ghdl-llvm package definition under guix for release 5.

We try to avoid having to propagate the whole clang suite to user path. Instead, we prefer to hard pointing to the correct binaries somewhere in the guix store.

I manage to build the package with no issue. As you can see, at run time I fall into an [error](https://codeberg.org/guix-science/guix-science/pulls/87#issuecomment-3028464) due to cc not found. Fair enough, the cc binary belongs to the guix store, and is not present in the user path. 

Now, the goal would be to patch ghdl to point to the right cc binary in the guix store. As I’m not familiar with ada, it is hard to guess where exactly. 

Any suggestion to replace the default cc search path by one specific one, somewhere in the system hierarchy ? 

Thanks,

tgingold, 2025-03-19:
You need to patch `ghdldrv.adb` (look for Linker_Cmd).

closed 2025-03-21 by tgingold

csantosb, 2025-03-21:
@tgingold Thanks for your help !