ghdl/ghdl #2934: crashed ghdl...

wm2015email, 2025-03-18:
ghdl -a --std=08 -frelaxed-rules -fsynopsys -fexplicit --warn-no-hide --no-vital-checks -Wno-shared --work=work test1.vhd

ghdl -e --std=08 -frelaxed-rules -fsynopsys -fexplicit --warn-no-hide --no-vital-checks -Wno-shared test_syntax

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (Ubuntu 1.0.0+dfsg-6) [Dunoon edition]
Compiled with GNAT Version: 10.3.1 20211117
Target: x86_64-linux-gnu
/mnt/c/Users/servi/OneDrive/Documents/GitHub/PerlEda/rmk_tests/
Command line:
/usr/bin/ghdl-mcode -e --std=08 -frelaxed-rules -fsynopsys -fexplicit --warn-no-hide --no-vital-checks -Wno-shared test_syntax
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1672
Call stack traceback locations:
0x5628ddc1f58a 0x5628dddbc4d0 0x5628dddbcfd6 0x5628dddbf005 0x5628dddbda28 0x5628dddbdc87 0x5628ddded8d3 0x5628dddee41b 0x5628dddedb99 0x5628dddef8ae 0x5628dddef968 0x5628dddf2790 0x5628dddf2a0f 0x5628dddf2b43 0x5628dde36881 0x5628dde34235 0x5628dddf2e01 0x5628dde1c3f1 0x5628dde1c747 0x5628dde1c846 0x5628dde23b61 0x5628dde4b806 0x5628dde097ef 0x5628dddfceac 0x5628dde0dd8c 0x5628dde080a0 0x5628dde4db55 0x5628ddd8eaf5 0x5628ddcc469e 0x5628dde514b3 0x5628dda7cb07 0x7ff567035d8e 0x7ff567035e3e 0x5628dda7b243 0xfffffffffffffffe
******************************************************************

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package my_types_pkg is
    type unsigned_vector is array (natural range <>) of unsigned;
    type slu_2dim is array (natural range <>, natural range <>) of std_logic;
    
    -- Conversion function declaration
    function ele_std_logic_2vector_to_slu_2dim(val: unsigned_vector; row: integer; col: integer) return slu_2dim;
end package;

package body my_types_pkg is
    function ele_std_logic_2vector_to_slu_2dim(val: unsigned_vector; row: integer; col: integer) return slu_2dim is
        variable result : slu_2dim(0 to row-1, 0 to col-1);
        variable idx    : integer := 0;
    begin
        assert (row * col = val'length) report "Dimension mismatch in std_logic_2vector_to_slu_2dim(). The product of rows and columns must equal the vector length." severity error;
        for i in 0 to row-1 loop
            for j in 0 to col-1 loop
                if (idx < val'length) then
                    result(i, j) := val(val'low + idx)(0);
                    idx := idx + 1;
                end if;
            end loop;
        end loop;
        return result;
    end function;
end package body;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.my_types_pkg.all;

entity test_syntax is
    -- Empty entity for testbench
end entity test_syntax;

architecture behavioral of test_syntax is
    -- Test constant declaration
    constant zeb_axi4_stream_crosspoint_axi4_xpoint_fifo_thresh : slu_2dim(0 to 13, 9 downto 0) := 
        ele_std_logic_2vector_to_slu_2dim(
            unsigned_vector'(
                0 => to_unsigned(0, 10),
                1 => to_unsigned(0, 10),
                2 => to_unsigned(0, 10),
                3 => to_unsigned(0, 10),
                4 => to_unsigned(0, 10),
                5 => to_unsigned(0, 10),
                6 => to_unsigned(0, 10),
                7 => to_unsigned(0, 10),
                8 => to_unsigned(0, 10),
                9 => to_unsigned(0, 10),
                10 => to_unsigned(0, 10),
                11 => to_unsigned(0, 10),
                12 => to_unsigned(0, 10),
                13 => to_unsigned(0, 10)
            ),
            14,  -- row
            10   -- col
        );
        
    -- Signal to observe constant in waveform
    signal test_output : std_logic := '0';
begin
    
    process
    begin
        -- Just a simple way to check that the syntax is valid
        -- Use the constant to make sure it's not optimized out
        test_output <= zeb_axi4_stream_crosspoint_axi4_xpoint_fifo_thresh(0, 0);
        wait for 10 ns;
        test_output <= zeb_axi4_stream_crosspoint_axi4_xpoint_fifo_thresh(13, 9);
        wait for 10 ns;
        
        report "Test completed successfully!";
        wait;
    end process;
    
end architecture behavioral;

wm2015email, 2025-03-18:
also: crashes:

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package my_types_pkg is
    type unsigned_vector is array (natural range <>) of unsigned;
    type slu_2dim is array (natural range <>, natural range <>) of std_logic;
    
    -- Conversion function declaration with simpler signature
    function ele_std_logic_2vector_to_slu_2dim(val: unsigned_vector; row: integer; col: integer) return slu_2dim;
end package;

package body my_types_pkg is
    function ele_std_logic_2vector_to_slu_2dim(val: unsigned_vector; row: integer; col: integer) return slu_2dim is
        variable result : slu_2dim(0 to row-1, 0 to col-1);
        variable idx    : integer := 0;
    begin
        -- Simplified assertion
        assert (row * col = val'length) report "Dimension mismatch" severity error;
        
        -- More straightforward loops
        for i in 0 to row-1 loop
            for j in 0 to col-1 loop
                if (idx < val'length) then
                    -- Extract bit 0 from each unsigned value
                    if val(idx)(0) = '1' then
                        result(i, j) := '1';
                    else
                        result(i, j) := '0';
                    end if;
                    idx := idx + 1;
                end if;
            end loop;
        end loop;
        return result;
    end function;
end package body;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.my_types_pkg.all;

entity test_syntax is
    -- Empty entity for testbench
end entity test_syntax;

architecture behavioral of test_syntax is
    -- Define the array separately for clarity
    constant MY_ARRAY : unsigned_vector(0 to 13) := (
        0 => to_unsigned(0, 10),
        1 => to_unsigned(0, 10),
        2 => to_unsigned(0, 10),
        3 => to_unsigned(0, 10),
        4 => to_unsigned(0, 10),
        5 => to_unsigned(0, 10),
        6 => to_unsigned(0, 10),
        7 => to_unsigned(0, 10),
        8 => to_unsigned(0, 10),
        9 => to_unsigned(0, 10),
        10 => to_unsigned(0, 10),
        11 => to_unsigned(0, 10),
        12 => to_unsigned(0, 10),
        13 => to_unsigned(0, 10)
    );
    
    -- Test constant declaration with separated expressions
    constant zeb_axi4_stream_crosspoint_axi4_xpoint_fifo_thresh : slu_2dim(0 to 13, 9 downto 0) := 
        ele_std_logic_2vector_to_slu_2dim(
            MY_ARRAY,
            14,  -- row
            10   -- col
        );
        
    -- Signal to observe constant in waveform
    signal test_output : std_logic := '0';
begin
    process
    begin
        -- Simple test to use the constant
        test_output <= zeb_axi4_stream_crosspoint_axi4_xpoint_fifo_thresh(0, 0);
        wait for 10 ns;
        test_output <= zeb_axi4_stream_crosspoint_axi4_xpoint_fifo_thresh(13, 9);
        wait for 10 ns;
        
        report "Test completed successfully!";
        wait;
    end process;
end architecture behavioral;

ghdl -a --std=08 -frelaxed-rules -fsynopsys -fexplicit --warn-no-hide --no-vital-checks -Wno-shared --work=work test1.vhd

ghdl -e --std=08 -frelaxed-rules -fsynopsys -fexplicit --warn-no-hide --no-vital-checks -Wno-shared test_syntax

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (Ubuntu 1.0.0+dfsg-6) [Dunoon edition]
Compiled with GNAT Version: 10.3.1 20211117
Target: x86_64-linux-gnu
/mnt/c/Users/servi/OneDrive/Documents/GitHub/PerlEda/rmk_tests/
Command line:
/usr/bin/ghdl-mcode -e --std=08 -frelaxed-rules -fsynopsys -fexplicit --warn-no-hide --no-vital-checks -Wno-shared test_syntax
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1672
Call stack traceback locations:
0x55844245b58a 0x5584425f84d0 0x5584425f8fd6 0x5584425fb005 0x5584425f9a28 0x5584425f9c87 0x5584426298d3 0x55844262a41b 0x558442629b99 0x55844262b8ae 0x55844262b968 0x5584426583da 0x558442658747 0x558442658846 0x55844265fb61 0x558442687806 0x5584426457ef 0x558442638eac 0x558442649d8c 0x5584426440a0 0x558442689b55 0x5584425caaf5 0x55844250069e 0x55844268d4b3 0x5584422b8b07 0x7f2a190ead8e 0x7f2a190eae3e 0x5584422b7243 0xfffffffffffffffe
******************************************************************

Paebbels, 2025-03-18:
Please update you GHDL installation. Your version is `1.0.0`. It's 4 years old and there have been hundreds of bug fixes and improvements.

See e.g. here for current releases as stable (v5.0.1) or nightly builds:  
https://github.com/ghdl/ghdl/releases

closed 2025-03-19 by tgingold