ghdl/ghdl #2938: A dedicated repository for GHDL issue tracking [Proposal]

Unike267, 2025-03-19:
Do you think it would be reasonable to create a new repository called `ghdl-issues`?

This repository would allow us to reproduce GHDL issues in a continuous integration (CI) environment, so we can track them simply by checking the code and the Actions results.

We could establish a systematic method, such as:

- Report the issue in the `ghdl/ghdl` repository.
- Create a directory in the root of the `ghdl/ghdl-issues` repository, named after the issue number, containing a minimal example to reproduce it.
- Add a CI job in `ghdl/ghdl-issues`, named `issue-number_nickname`, to run the reproduction test.
- If someone proposes a fix, they create a subdirectory inside the issue directory, named `fixing-attempt_fixer-nickname`.
- Add a new CI job named `issue-number_fixing-attempt_fixer-nickname` to test the proposed fix.

### Example: Issue #2930

- Report issue 2930 in `ghdl/ghdl`.
- Create a directory `2930` in `ghdl/ghdl-issues`, containing the source code initially shared in the issue discussion.
- Add a CI job named `2930_marzoul`.
- If I want to attempt a fix, I create a subdirectory `2930/1_unike267` where I upload additional test data (e.g., a 150MB hex file, which cannot be attached to the issue discussion) and my modifications of the initial code.
- Create a new CI job named `2930_1_unike267` to run my proposal.

Would this approach make sense?

/cc @tgingold @Paebbels @umarcor

Paebbels, 2025-03-19:
I don't think so. GHDL issues get usually closed in a few days and in many days within 1 week.

There are some longer lasting issues, but these are for new features and especially features requiring conceptional/architecture changes in GHDL itself.

@umarcor made once a system where reported code snippets within an issue can be checked by CI. See the issue template.

Unike267, 2025-03-19:
This feature has enormous potential. See https://github.com/umarcor/issue-runner/tree/main

Currently, the code blocks pasted in an issue are static (they don’t trigger any CI job). The idea behind this project is to parse the issue’s markdown, detect VHDL code blocks, extract them, and automatically generate a CI test job.

Then, the feature would reply with a comment to the original issue (including CI results/outcome), while also tagging the issue.

This is an incredible idea. It’s a real pity that it hasn’t been implemented yet.

We should definitely free up some time to work on it.

Just my opinion.

Unike267, 2025-03-20:
Hi!

I'm very busy with my thesis but really I would like to tackle this problem.

So I will try something!

(No rush, don't expect anything in the next few months.)

> There are some longer lasting issues, but these are for new features and especially features requiring conceptional/architecture changes in GHDL itself.

Let's let this be one of those.

Unike267, 2025-03-24:
## Update

See: https://github.com/Unike267/2938/actions/runs/14042929230/job/39317140064

Job triggered by: https://github.com/Unike267/2938/issues/1

Unike267, 2025-03-28:
## Update

Check: https://github.com/Unike267/2938/actions/runs/14128555659/job/39583143862

Job triggered by: https://github.com/Unike267/2938/issues/7

---

I've added an ISSUE Template that automatically assigns the CIFERIS label through this header:

```
---
name: Bug report with CI
about: This issue operates with the CIFERIS-(CI) o(f) the issu(e) trigge(r)ed by the (is)sue feature integrated
title: ''
labels: CIFERIS, bug
assignees: ''

---
```

Then, I modified the CIFERIS YAML by adding `if: contains(github.event.issue.labels.*.name, 'CIFERIS')` to the jobs.

(See https://github.com/Unike267/2938/blob/main/.github/workflows/CIFERIS.yml)

In this way, only the issues labelled with "CIFERIS" will trigger the CI. 

Therefore, if you chose the "Bug report with CI" template, the "CIFERIS" label will be added automatically (it must be created beforehand) and the CI it will run.

Any feedback?

Unike267, 2025-03-28:
Interesting result with: #2953 

Emulated by: https://github.com/Unike267/2938/issues/8

See https://github.com/Unike267/2938/actions/runs/14135163046/job/39605047310 (OK) VS https://github.com/Unike267/2938/actions/runs/14135163046/job/39605047304 (FAIL)

It seems that something have been scanned either the top, or the name of the HDL to correctly name the extracted block (Unless the use of `*.vhd` is normalized) 🤔

tgingold, 2025-03-29:
@Unike267 I think this would be very useful to ensure the mwe is present and complete.

However, how do you distinguish the right mwe which triggers a bug and an mwe with a stupid syntax error ?  That will be on the user.  Fine.

Paebbels, 2025-03-29:
I remember on of @umarcor's goals was to repeatedly check, if a new change in GHDL has also fixed an older issue or similar/duplicate issue. So once the MWE is correctly setup and not a syntax error (we would close such issues), we have somehow a regression on old issues in the backlog.

Unike267, 2025-03-29:
> That will be on the user. Fine.

Yeah that's is the idea. 

However, if the user makes an evident syntax error and one of us realizes (seeing the code/CI) we could answer to the user and when he/she edits the issue with the right mwe, the editing action will relaunch the CI.

I understand that the idea is to do this automatically.

Now cold I can think of something like adding a new job that processes the ghdl result and with the command `gh issue edit --add-label $processing_result` label the issue.

The option of the variable ` $processing_result` could be:

- mwe reproducible
- mwe not reproducible...

However, CI must be run (Bug with CI template triggers CI always) to know these results 🤔.

Unike267, 2025-03-29:
>  if a new change in GHDL has also fixed an older issue or similar/duplicate issue

It occurs to me adding to $processing_result variable

> The option of the variable  $processing_result could be:

the `pending bug` value to label the issue with this value if for example the ghdl result is:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 6.0.0-dev (b6.0.0.r42.g5d39d5095) [Dunoon edition]
[...]
******************************************************************
```

Thus, when a change in ghdl code is made trigger only the CI of the issues labeled with this value, and close them through `gh issue close` if the CI doesn't broke. 🤔