ghdl/ghdl #2942: Missing escape for coverage JSON

cwyki, 2025-03-21:
**Description**
When calling ghdl with windows pathes (including backslash), the coverage JSON output contains the file paths without escaped backslashes, making it invalid JSON. The `ghdl coverage` tool though does not seem to have issues with it.

**Expected behaviour**
GHDL outputs a valid JSON with escaped special characters (like `\`).

**How to reproduce?**

```vhd :file: reproducer/json_format.vhd
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;

ENTITY json_format IS
END ENTITY json_format;

ARCHITECTURE rtl OF json_format IS
    SIGNAL clk : std_logic := '0';
BEGIN
    clk <= '1';

    main : PROCESS
    BEGIN
        WAIT;
    END PROCESS;

END ARCHITECTURE;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --work=lib --std=08 reproducer\json_format.vhd
ghdl --elab-run --work=lib --std=08  --coverage --coverage-output=reproducer.json json_format
ghdl coverage --format=lcov reproducer.json
more reproducer.json
```
results in
```json
{
 "app": "GHDL coverage output file",
 "version": "1.0.0",
 "testcase": "unknown",
 "timestamp": "20250321113309.137",
 "outputs": [
  {
   "file": "reproducer\json_format.vhd",
   "dir": ".",
   "sha1": "4597ef6c3ea468cae2dbd1e6645050d5f1810eb3",
   "mode": "stmt",
   "max-line": 14,
   "result": {
    "10": 1,
    "14": 1
   }
  }
 ]
}

```
where the `\j` in the "file" value should read `\\j`.

**Context**

- OS: Windows 11
- Origin:
  - [ ] Package manager:
  - [x] Released binaries: GHDL 5.0.1
  - [ ] Built from sources:

**Additional context**
```
GHDL 5.0.1 (4.1.0.r602.g37ad91899.dirty) [Dunoon edition]
 Compiled with GNAT Version: 14.2.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2025 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Paebbels, 2025-03-21:
The field `timestamp` is currently formatted as a string (`"20250321113309.137"`), but should be formatted as a floating point number: `20250321113309.137`.

closed 2025-03-29 by tgingold

Paebbels, 2025-03-29:
@tgingold does it also improve the timestamp format to the ISO format?

tgingold, 2025-03-29:
No, but I don't think this is needed.

Paebbels, 2025-03-29:
I don't think we should introduce user defined time formats, when there is the ISO 8601 format widely used in YAML and JSON formats. Even without a conversion function, it can easily formatted by adding delimiters into the concatenated string.

Also on the reader side, it doesn't require a custom regexp to split the string and special code for validation (e.g. in range), because a commonly available `fromiso()` method or function can be used by any language parsing this timestamp.

The current format with a dot suggests it's a double value, but it's not. It's a concatenation of multiple integer values as a string. Please let use use ISO 8601 format. The timezone is not mandatory, but when collecting evidence for regression testing, reports without a timezone are not good in a globally interconnected world, when CI serves run with a local time. Thus ordering of reports not ensured.

tgingold, 2025-03-31:
This is not a file for the user, but an internal file that contains raw results before they are merged.

Paebbels, 2025-03-31:
I think all data formats - also internal formats - should use proper formats and datatypes.

What is the format after merging the coverage data?  
Do I need to run GHDL to merge it or can it be done by other tools too?
Is there any data lost when merging to another format?

------
[pyEDAA.Reports](https://github.com/edaa-org/pyEDAA.Reports) can read multiple report formats. For now it's mainly "JUnit XML" dialects, but it can also read first parts of Cobertura and display code coverage in [Sphinx documentation](https://github.com/pyTooling/sphinx-reports). Besides code coverage, it can also display documentation coverage in Sphinx.

I'm interested in also integrating GHDLs coverage results into that system and being able to integrate reports into my Sphinx documentation.

commit f2435196 (2025-03-29): ghdldrv: quote and unquote backslash in json files

Fix #2942