ghdl/ghdl #2943: GHDL Crashes When Loading a Large HEX File

Unike267, 2025-03-21:
**Description**
The running process crashes while loading a very large file.

**Expected behaviour**
The process should either complete successfully or provide a GHDL-specific error message instead of crashing at the system level.

**How to reproduce?**

**HDL files**

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.env.stop;

library MEM_MGMT;
use MEM_MGMT.MEM_package.all;

entity test is

end test;

architecture tb of test is

  constant clk_period : time      := 10 ns;
  signal   clk        : std_logic := '0';

  signal ren   : std_logic := '0';
  signal wen   : std_logic := '0';
  signal waddr : std_logic_vector(31 downto 0) := (others => '0');
  signal wdata : std_logic_vector(31 downto 0) := (others => '0');
  signal raddr : std_logic_vector(31 downto 0) := (others => '0');
  signal rdata : std_logic_vector(31 downto 0) := (others => '0');

begin

  uut : 
  entity MEM_MGMT.top
                     generic map(
                                DATA_BITS   => 32,
                                CELLS => 2 ** 24,
                                ADDR_BITS => 32
                                )
                     port    map(
                                clk   => clk,
                                wen   => wen,
                                waddr => waddr,
                                wdata => wdata,
                                ren   => ren,
                                raddr => raddr,
                                rdata => rdata
                                );

  clk <= not clk after clk_period/2;

  read_value : process (clk) is
    begin
      if (rising_edge(clk)) then
         ren <= '1';
         raddr <= std_logic_vector(to_unsigned(8388607, 32));
      end if;
    end process read_value;

  testing : process is
  begin
    wait for 2*clk_period;
    -- For this address the output must match 0x8dc3284f (2378377295)
    report "The position eight million three hundred eighty-eight thousand six hundred eight (in decimal) is: " & integer'image(to_integer(unsigned(rdata)));
    stop;
  end process testing;

end tb;
```

```vhd :file: top.vhd

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

 -- Change from MEM to MEM_MGMT (memory management) for not hide library "mem" with declaration of "mem" signal
 library MEM_MGMT;
 use MEM_MGMT.MEM_package.all;

entity top is
	generic (
		DATA_BITS : natural := 32;
		CELLS     : natural := 2 ** 24;
		ADDR_BITS : natural := 32
	);
	port (
		clk   : in  std_logic;
		wen   : in  std_logic;
		waddr : in  std_logic_vector(ADDR_BITS-1 downto 0);
		wdata : in  std_logic_vector(DATA_BITS-1 downto 0);
		ren   : in  std_logic;
		raddr : in  std_logic_vector(ADDR_BITS-1 downto 0);
		rdata : out std_logic_vector(DATA_BITS-1 downto 0)
	);
end top;

architecture synth of top is

    constant ROM_LOAD_FILE_32 : string := "string.hex";
    signal mem :t_MEM32 (0 to CELLS-1) := MEM32_INIT_HEX (ROM_LOAD_FILE_32,  CELLS);

	attribute ram_style : string;
	attribute ram_style of mem : signal is "block";

begin

	process(clk)
	begin
		if rising_edge(clk) then

			if ren = '1' then
				rdata <= mem(to_integer(unsigned(raddr)));
			end if;

			if wen = '1' then
				mem(to_integer(unsigned(waddr))) <= wdata;
			end if;

		end if;
	end process;

end architecture;
```

```vhd :file: MEM_package.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;

package MEM_package is

  type t_MEM32 is array (natural range <>) of std_logic_vector(31 downto 0); -- Memory with 32-bit data width

  impure function MEM32_INIT_HEX(LOAD_FILE : string; MEM_ITEMS : natural) return t_MEM32;

end MEM_package;

package body MEM_package is

  impure function MEM32_INIT_HEX(LOAD_FILE : STRING; MEM_ITEMS : natural) return t_MEM32 is
    file     mem_file  : text open read_mode is LOAD_FILE;
    variable mem_line  : line;
    variable temp_word : std_logic_vector(31 downto 0);
    variable temp_mem  : t_MEM32(0 to MEM_ITEMS-1) := (others => (others => '0'));
  begin
    for i in 0 to MEM_ITEMS-1 loop
      exit when endfile (mem_file);
      readline(mem_file, mem_line);
      hread  (mem_line, temp_word);
      temp_mem(i)     := temp_word;
    end loop;
    return temp_mem;
  end function;

end MEM_package;
```

**Containerfile:**

To generate an image with the latest version of GHDL, I verified this issue using the repository at commit `7a954fd3cc9ddc8b54b1b990ec1d2c51a4aa4f47`.

```dockerfile :img: ghdl.containerfile
# Licensed under the GNU General Public License v3.0;
# You may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     https://www.gnu.org/licenses/gpl-3.0.html

# GHDL container

MAINTAINER <unike267@gmail.com>

FROM ubuntu:latest AS base

ARG GNAT_VER="13"
ARG LLVM_VER="18"

# Install dependencies
RUN apt-get update -qq \
 && DEBIAN_FRONTEND=noninteractive apt-get -y install --no-install-recommends \
    ca-certificates \
    clang-$LLVM_VER \
    gcc \
    gnat-$GNAT_VER \
    llvm-$LLVM_VER-dev \
    make \
    zlib1g-dev \
 && apt-get autoclean && apt-get clean && apt-get -y autoremove \
 && update-ca-certificates \
 && rm -rf /var/lib/apt/lists/* 

FROM base AS build

ARG LLVM_VER="18"

# Install ghdl
RUN apt-get update -qq \
 && DEBIAN_FRONTEND=noninteractive apt-get -y install --no-install-recommends \
    git \
 && git clone https://github.com/ghdl/ghdl \
 && cd ghdl \
 && mkdir build-llvm \
 && cd build-llvm \
 && CXX=clang++-$LLVM_VER ../configure --with-llvm-config=llvm-config-$LLVM_VER --default-pic --disable-werror \
 && make -j$(nproc) \
 && make DESTDIR=/opt/ghdl/ install \
 && cd ../.. \
 && rm -rf ghdl 

# Temporary scratch image for “transporting” the compiled GHDL binaries
FROM scratch AS tmp_ghdl
COPY --from=build /opt/ghdl/ /ghdl/

# Final ghdl dependent target
FROM base AS ghdl
COPY --from=tmp_ghdl /ghdl/usr/local /usr/local/

RUN ghdl --version
```

**Setup to reproduce**

(Feel free to skip any steps if they are redundant for you.)

```sh :image: CUSTOM-IMAGE
apt-get -y install podman
podman build -f  ghdl.containerfile -t ghdl:7a954fd --target ghdl
mkdir 2943
cd 2943
mkdir build
curl -L -o string.hex https://www.nodocracy.com/ghdl-2943/string.hex
podman run --rm -itv $(pwd):/wrk:Z -w /wrk ghdl:7a954fd ghdl -i --std=08 --workdir=build --work=MEM_MGMT *.vhd
podman run --rm -itv $(pwd):/wrk:Z -w /wrk ghdl:7a954fd ghdl -m --std=08 --workdir=build --work=MEM_MGMT test
podman run --rm -itv $(pwd):/wrk:Z -w /wrk ghdl:7a954fd ghdl -r --std=08 --workdir=build --work=MEM_MGMT test --max-stack-alloc=524288
```

**Additional Files**

I attempted to compress the file into a tarball/zip, but both were too large to upload.
For now, I have temporarily uploaded the HEX file to my webpage.

The file contains a hex array generated from random 32-bit data. It has `2**24` elements, so I recommend using an efficient editor like Emacs to open/edit it.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`
  - [x] Custom Container (I make an image and do not use one already made to take into account the last commits that Tristan has made regarding efficiency improvement.)

**Additional context**

This issue was opened based on Tristan's comment:

> @Unike267 If this is a crash, please open a new issue.

System info:

[image]

tgingold, 2025-03-23:
Works for me after a few changes:
1) Don't use signals for memory, uses variables.  std_logic requires 1 byte when it's a variable, but ~128B when it's a signal.
   Your program requires several GB of memories
2) Use 'ulimit -s 600000' to increase stack size
3) Use `to_hstring` to display the value in `testing`.  The unsigned doesn't fit in an integer.

Paebbels, 2025-03-23:
Dual port BlockRAM (36kBit) or UltraRAM (288kBit) are usually designed as signals used by a process using 1 or 2 clock signals in a process.

Using a shared variable has several drawbacks:
1. It's not VHDL-2008 compliant to use ordinary types for shared variables and a two process implementation.
2. Variables can't be debugged by many tools (xSim, GHDL).  
   Memory content is not visible in the waveform.

-----
Why is the ratio between a std_logic as variable vs. signal so huge?

* 1B: current value
* 1B: last value
* 8B: last change
* ...

tgingold, 2025-03-24:
You don't need a shared variable.  A process variable is ok.

You need also memory for last_event, drivers, sensitivity, scheduling...

Unike267, 2025-03-24:
> Works for me after a few changes:
Don't use signals for memory, uses variables. std_logic requires 1 byte when it's a variable, but ~128B when it's a signal.
Your program requires several GB of memories
Use 'ulimit -s 600000' to increase stack size
Use to_hstring to display the value in testing. The unsigned doesn't fit in an integer.

This has also worked for me!

Let me present how I have applied your proposed changes:

```diff
-    signal mem :t_MEM32 (0 to CELLS-1) := MEM32_INIT_HEX (ROM_LOAD_FILE_32,  CELLS);

-	attribute ram_style : string;
-	attribute ram_style of mem : signal is "block";

begin

	process(clk)
+         variable mem :t_MEM32 (0 to CELLS-1) := MEM32_INIT_HEX (ROM_LOAD_FILE_32,  CELLS);
+	  attribute ram_style : string;
+	  attribute ram_style of mem : variable is "block";
	begin
		if rising_edge(clk) then

			if ren = '1' then
				rdata <= mem(to_integer(unsigned(raddr)));
			end if;

			if wen = '1' then
-				mem(to_integer(unsigned(waddr))) <= wdata;
+ 				mem(to_integer(unsigned(waddr))) := wdata;
			end if;
```

```diff
-    report "The position eight million three hundred eighty-eight thousand six hundred eight (in decimal) is: " & integer'image(to_integer(unsigned(rdata)));
+    report "The position eight million three hundred eighty-eight thousand six hundred eight (in hex) is: " & to_hstring(rdata);
```

```diff
- podman run --rm -itv $(pwd):/wrk:Z -w /wrk ghdl:7a954fd ghdl -r --std=08 --workdir=build --work=MEM_MGMT test --max-stack-alloc=524288
+ podman run --rm -itv $(pwd):/wrk:Z -w /wrk ghdl:7a954fd sh -c "ulimit -s 600000 && ghdl -r --std=08 --workdir=build --work=MEM_MGMT test --max-stack-alloc=524288"
```

Functional test:

```
usainz@U013922:~/Escritorio/tmp/2943/Tristan-correction$ podman run --rm -itv $(pwd):/wrk:Z -w /wrk ghdl:7a954fd sh -c "ulimit -s 600000 && ghdl -r --std=08 --workdir=build --work=MEM_MGMT test --max-stack-alloc=524288" 
test.vhd:58:5:@20ns:(report note): The position eight million three hundred eighty-eight thousand six hundred eight (in hex) is: 8DC3284F
simulation stopped @20ns
```

FTR, on my computer the process took 39 seconds to complete. 

Maybe someone would like to answer this interesting question:

> Why is the ratio between a std_logic as variable vs. signal so huge?

If not, the issue is concluded for me. 😄

tgingold, 2025-03-26:
The ratio is large because for signals you need to track metadata: last value, last event, scheduled processes, drivers...

closed 2025-03-26 by tgingold