ghdl/ghdl #2945: generate else generate : exception raised: unknown reason

NicoPy, 2025-03-23:
Hi,

Using `generate else generate` in VHDL-2008 gives an error : `exception raised: unknown reason`

The test code :
```vhdl
library ieee; 
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test_generate is
end entity test_generate;

architecture rtl of test_generate is
    
    constant VERSION_A : boolean := False;
    
    signal version : std_logic_vector(7 downto 0);
    
begin
    
    g_version:
    if VERSION_A generate
        version <= x"10";
    else generate
        version <= x"20";
    end generate;
    
end rtl;
```

The commands to run the simulation :
```
ghdl -a  --std=08  --workdir=sim_build --work=work ./tb_generate.vhd
ghdl -r  --std=08  --workdir=sim_build --work=work -Psim_build  test_generate --wave=test_generate_waveforms.ghw
```

Using 2 `generate` statements with complementary conditions like what is needed with VHDL-93 is OK.

Paebbels, 2025-03-23:
Please add the full exception text.

tgingold, 2025-03-23:
Also, try with `--fst=` instead of `--wave=` (or without `--wave=`).
The support of `--wave=` is not fully complete with vhdl 2008.

Paebbels, 2025-03-23:
@tgingold do we have a similar issue with `case .. generate`?

closed 2025-03-29 by tgingold

tgingold, 2025-04-11:
The `--fst=` dumper has been enhanced, and the `ghw` one still has to be.
I don't think FST is currently degraded compared to GHW.

NicoPy, 2025-04-11:
Thanks for your answer.
Which file format do you recommend to use ?

tgingold, 2025-04-12:
Currently I recommend FST with the mcode backend.

NicoPy, 2025-04-12:
OK. Thanks for your support.