ghdl/ghdl #2946: request for stack dump for vhdl library frunctions that squawk too much....

wm2015email, 2025-03-24:
When i run the code below ghdl  doesn't report a line number that i can fix...  it just reports truncation...  I realize this is because the vhdl standard people never fix this problem in vhdl of the libraries squawking about an error without reporting the line location in the code where its called... but is there something ghdl can do to help people find the line with the "truncation" without needing to keep deleting code to locate it?  like maybe dump the calling stack of the to_ufixed function instead of just this:

../../src/ieee2008/fixed_generic_pkg-body.vhdl:2484:9:@0ms:(assertion warning): :ieee:fixed_generic_pkg:TO_UFIXED(NATURAL): vector truncated
  from: ieee.fixed_generic_pkg.to_ufixed at fixed_generic_pkg-body.vhdl:2484

so you can tell where in the code to_ufixed is being used.

```
library ieee;

use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;
use ieee.fixed_pkg.all;
use ieee.fixed_float_types.all;

entity j_csr is
port (
    csr_add_noise_sig_gain          :out  ufixed(7 downto -10)
);
end entity;

architecture rtl of j_csr is

    constant zeb_add_noise_sig_gain_17_0         :ufixed(7 downto -10)           := to_ufixed(1515520, 7, -10);

begin

end architecture;
```

closed 2025-03-25 by tgingold

tgingold, 2025-03-25:
@Unike267 Update and try again...

Unike267, 2025-03-25:
It works now!

Output:

```
ghdl:note: top entity is "j_csr"
../../src/ieee2008/fixed_generic_pkg-body.vhdl:2484:9:(assert warning): :ieee:fixed_generic_pkg:TO_UFIXED(NATURAL): vector truncated
 called from: architecture "rtl" of "j_csr" at j_csr.vhd:15:14
```

Unike267, 2025-03-25:
> ghdl:note: top entity is "j_csr"

Btw Tristan, What is the scope of the GHDL in order to autocoplete the name of the top? Only with mcode, synth...? 🤔

tgingold, 2025-03-26:
I think finding automatically the top unit is supported by synth and command `-c` (for jit backends).

BTW, this new backtrace is available only during elaboration. 
I think we also need to tune the conditions to display it.

commit 6240d7d4 (2025-03-25): synth-vhdl_stmts: disp backtrace on failed assertion

Fix #2946