ghdl/ghdl #2950: IF anybody sees the IEEE VHDL standard people tell them they need a bitwise cast syntax

wm2015email, 2025-03-27:
If the type is the same, example:

std_logic_vector   array (natural range <>) of std_logic;
signed             array (natural range <>) of std_logic;
unsigned        array (natural range <>) of std_logic;
ufixed             array (integer range <>) of std_logic;
sfixed              vhdl 2008 array (integer range <>) of std_logic;

it would be extremely nice to have a bitwise cast that is exactly the same for all types of the same type definition that will work when assigning a component port from an unsigned to a std_logic_vector...FPGA tools already do this automatically anyway when passing signal to systemverilog or when connecting the toplevel io... why can't VHDL do it internally as well...   seem like a gross oversight in the VHDL standard for year...  I actually want to remove VHDL entities and replace them with verilog just to avoid the type conversion ugliness...  this is what makes people throw away vhdl in favor verilog in my option.... too much trouble to do what takes no effort in verilog...

my_Component: entity work.my_entity (
     my_std_logic_vector_output => my_unsigned_input   -- too annoyinhg where is the bitwise case operator that works here? this is where i'm goign to delete vhdl and repladce it with verilog befcause it automatically converts it...
)

wm2015email, 2025-03-27:
my_Component: entity work.my_entity (
my_std_logic_vector_output -> my_unsigned_input
);

maybe use a "->" operator to force bitwise type conversion , compared to normal type safe connetion? "=>"

xyz <= abd;  type safe
xyz <- abd;   bitwise cast assignement that works like verilog... or with types that have the same types definition...

wm2015email, 2025-03-27:
xyz := abd; type safe variable assign
xyz :- abd;  bitwise cast assigment matchig with verilog...

Paebbels, 2025-03-28:
@wm2015email GHDL is not the right place to report and complain about the VHDL standard. Please report at https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues

And yes, me and others here are also involved in the IEEE SA for P1076.

Please also give a detailed description of your problem and what you want to achieve, so we can discuss your request.

----------
As a general answer:  
I looks like your missing basic understanding of why programming languages have strong typing and need casting. This is not just VHDL, but also C, C++, Java, Visual Basic .NET, C#, ... And even dynamic programming languages like JavaScript and Python figured out, that a language without type checking can't be used sufficiently in large projects, so they invented TypeScript and Python Type Annotations.

In VHDL, everything is explicit and not implicit, so the engineer ensures to the compiler he has understood the implications and knows what he's doing. This ensures for save and reliable hardware designs. Please also have a look what industries/markets use VHDL and which use SystemVerilog ...

Because Systemverilog has implicit operations and less checks from the compiler, the user needs to write more tests and checks to compensate the lack of strictness from the compiler.

We're open to discuss your improvement suggestions at [VHDL-Issues](https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues). Please report over there.

closed 2025-03-28 by Paebbels