ghdl/ghdl #2951: GHDL raise an error when to_hstring is called with an empty std_ulogic_vector

pacalet, 2025-03-28:
**Description**
GHDL raise an error when `to_hstring` is called with an empty `std_ulogic_vector`. It does not raise an error with an empty `bit_vector`. This is due to the way `to_hstring` is written in `lib/ghdl/src/ieee2008/std_logic_1164-body.vhdl` at line 1059: `if value (value'left) = 'Z' then`. A test on `value'length` is missing.

**Expected behaviour**
An empty string should be returned.

**How to reproduce?**

```vhd :file: foo.vhd
use std.textio.all;

library ieee;
use ieee.std_logic_1164.all;

entity foo is end entity foo;

architecture bar of foo is
begin
  process
    variable s: string(2 to 1);
    variable v1: bit_vector(-1 downto 0);
    variable v2: std_ulogic_vector(-1 downto 0);
  begin
    s := to_hstring(v1);
    report "bit_vector OK";
    s := to_hstring(v2);
    report "std_ulogic_vector OK";
    wait;
  end process;
end architecture bar;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 foo.vhd
ghdl -r --std=08 foo
```
**Context**
Please, provide the following information:

- OS: macOS 13.3.1 (22E261)
- Origin:
  - [ ] Package manager: `macports 2.10.5, `
  - [X] Released binaries: `https://github.com/ghdl/ghdl/releases/download/v5.0.1/ghdl-mcode-5.0.1-macos13-x86_64.tar.gz`
  - [ ] Built from sources: `commit SHA`

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

Paebbels, 2025-03-28:
Please report this problem at https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues so the fix can be discussed for the next IEEE Std. P1076 release.

pacalet, 2025-03-28:
There is already a 2 years old open issue about this, there. But you're right, if this VHDL code comes from there it is where it should be fixed. I submitted a pull request. Feel free to reject if using the same source code as IEEE-P1076 is more important than fixing this bug.

Paebbels, 2025-03-28:
So you refer to https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/287, right?

----
See also: https://github.com/ghdl/ghdl/issues/2338

pacalet, 2025-03-28:
Yes.

pacalet, 2025-03-31:
Just tested with ModelSim SE-64 2021.3 (Jul 13 2021 Linux 6.1.0-17-amd64):

```
VSIM 1> run -all
# ** Note: bit_vector OK
#    Time: 0 ns  Iteration: 0  Instance: /foo
# ** Note: std_ulogic_vector OK
#    Time: 0 ns  Iteration: 0  Instance: /foo
```

So, apparently it has been fixed. However the source code of `std_logic_1164` that can be found in the install tree is still the same and has the bug... What they probably did is to ship a fixed compiled version while keeping the original source version.