ghdl/ghdl #2955: VHDL Synthesis: error array port fully or partially unconstrained

sideprojectslab, 2025-03-29:
**Description**
VHDL-2008, when an entity has an input port of "array type" with more than one dimension, such as "signed_vector", the analyze/elaborate step fails if any of the port dimensions are unconstrained.

```
data_in : in  signed_vector;  -- not OK
data_in : in  signed_vector(0 to 7) ; -- not OK
data_in : in  signed_vector(0 to 7)(31 downto 0); -- OK
```

the reported error is:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.1 (4.1.0.r602.g37ad91899.dirty) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-w64-mingw32
C:\Local\Data\eas\projects\eas\eas_ip_lib\svn\branches\release\v2\development\eas_ip_lib\fpga\vhdl\dsp\misc\test\test_sw_soft_mute\ghdl\_make\
Command line:
C:\Local\Development\compilers\GHDL 5.0\bin\ghdl.exe -r --std=08 --workdir=work -frelaxed aaa_test_tle --fst=waves.fst --max-stack-alloc=1024000
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : synth-vhdl_expr.adb:635
******************************************************************
```

**Expected behaviour**
all the three port definitions above should be valid VHDL-2008

**How to reproduce?**

define an unconstrained multi-dimensional array type in some package:

`type signed_vector   is array (natural range <>) of signed;`

make a dummy entity that uses the type as a port, only partially constrained (it would normally be constrained when the module is instantiated and a signal is connected to it) and observe that the ananlysis/elaboration step fails.

```
entity dummy is
	generic
	(
		g_bit_width         : positive;
	);
	port
	(
		i_data : in  signed_vector(0 to g_num_chnl - 1);
		o_data : out signed_vector(0 to g_num_chnl - 1)(g_bit_width - 1 downto 0)
	);
```

Paebbels, 2025-03-29:
@sideprojectslab to be correct, this is not a multi-dimensional array, it's just a nested array in an array.

Nested 1 dimensional array in an other 1 dimensional array:
```vhdl
type std_logic_vector_vector is array(natural range <>) of std_logic_vector;

signal outer_array : std_logic_vector_vector(0 to 3)(7 downto 0);
inner_array <= outer_array(2);
element <= outer_array(2)(6)
```

2 dimensional array:
```vhdl
type std_logic_matrix is array(natural range <>, natural range <>) of std_logic;

signal matrix : std_logic_matrix(0 to 3, 7 downto 0);
element <= matrix(2, 6)
```

You can setup more then 2 dimensions. Array attributes like `'range(2)` accept a dimension parameter, default is 1. Of cause you can have other composite types (array, record) as elements of such multi dimensional arrays.

--------------
Maybe you can extend your example, and we check if this kind of true multi-dimensional array is problematic too.

sideprojectslab, 2025-03-29:
> [@sideprojectslab](https://github.com/sideprojectslab) to be correct, this is not a multi-dimensional array, it's just a nested array in an array.
> 
> Nested 1 dimensional array in an other 1 dimensional array:
> 
> type std_logic_vector_vector is array(natural range <>) of std_logic_vector;
> 
> signal outer_array : std_logic_vector_vector(0 to 3)(7 downto 0);
> inner_array <= outer_array(2);
> element <= outer_array(2)(6)
> 
> 2 dimensional array:
> 
> type std_logic_matrix is array(natural range <>, natural range <>) of std_logic;
> 
> signal matrix : std_logic_matrix(0 to 3, 7 downto 0);
> element <= matrix(2, 6)
> 
> You can setup more then 2 dimensions. Array attributes like `'range(2)` accept a dimension parameter, default is 1. Of cause you can have other composite types (array, record) as elements of such multi dimensional arrays.
> 
> Maybe you can extend your example, and we check if this kind of true multi-dimensional array is problematic too.

Thanks for the clarification, and thanks for teaching me something today :D I have edited the original text to be more accurate

tgingold, 2025-03-29:
Please, provide a reproducer.

sideprojectslab, 2025-03-30:
Here you go. This is the version that fails, notice that the array port is being indexed in this case, so the port constraint inference is not super obvious. Further down I added a version without indexing that compiles fine:

```
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;
use     ieee.math_real.all;

package dummy_pkg is
	type signed_vector is array(natural range <>) of signed;
end package;

library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;
use     ieee.math_real.all;

library work;
use     work.dummy_pkg.all;

entity dummy is
	generic
	(
		g_bit_width : positive;
		g_num_chnl  : positive
	);
	port
	(
		i_data : in  signed_vector(0 to g_num_chnl - 1);
		o_data : out signed_vector(0 to g_num_chnl - 1)(g_bit_width - 1 downto 0)
	);
end entity;

architecture rtl of dummy is
begin
	o_data <= i_data;
end architecture;

library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;
use     ieee.math_real.all;

library work;
use     work.dummy_pkg.all;

entity aaa_test_tle is
end entity;

architecture rtl of aaa_test_tle is

	constant c_bit_width : positive := 16;
	constant c_num_chnl  : positive := 1;
	signal i_data        : signed(c_bit_width - 1 downto 0) := (others => '0');
	signal o_data        : signed(c_bit_width - 1 downto 0);

begin

	i_dut : entity work.dummy
	generic map
	(
		g_bit_width => c_bit_width,
		g_num_chnl  => c_num_chnl
	)
	port map
	(
		i_data(0)   => i_data,
		o_data(0)   => o_data
	);

end architecture;
```

Here a version that passes (just the tle architecture, the rest stays the same:

```
architecture rtl of aaa_test_tle is

	constant c_bit_width : positive := 16;
	constant c_num_chnl  : positive := 1;
	signal i_data        : signed_vector(0 to c_num_chnl - 1)(c_bit_width - 1 downto 0);
	signal o_data        : signed_vector(0 to c_num_chnl - 1)(c_bit_width - 1 downto 0);

begin

	i_dut : entity work.dummy
	generic map
	(
		g_bit_width => c_bit_width,
		g_num_chnl  => c_num_chnl
	)
	port map
	(
		i_data   => i_data,
		o_data   => o_data
	);

end architecture;
```

I hope this helps. The first architecture does compile fine in Modelsim and Riviera, as the input port is in fact fully constrained, though I am not 100% sure it is compliant with the standard.

dalex78, 2026-08-20:
# Issue #2955 — Crash: nested-array port fully or partially unconstrained

Analysis by an AI agent.

## High-level summary

An entity port of a nested array type (`signed_vector`, an array of unconstrained `signed`) left fully unconstrained (`i_data : in signed_vector;`) or partially constrained (only the outer dimension: `signed_vector(0 to N-1)`, elements still unconstrained) crashed analysis/elaboration with an internal `ASSERTION_ERROR` at `synth-vhdl_expr.adb:635`. Fully constraining both dimensions (`signed_vector(0 to N-1)(W-1 downto 0)`) was reported as working.

## Investigation / current status: already fixed on master

The report's reproducer was incomplete (referenced an undeclared `g_num_chnl` generic); reconstructed it faithfully, adding the missing generic, and instantiated it from a top-level with a fully-constrained actual signal (as the report describes: "assign a fully constrained signal to both i_data and o_data"). Tested both failing shapes from the report — a fully unconstrained port and an outer-only-constrained port — against current `master`: both analyze and simulate cleanly, no crash. This must have been fixed as part of general improvements to nested-unconstrained-array port handling since this was reported (2025). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/gna/issue2955/` (a faithful reconstruction of the report's repro, covering both the fully-unconstrained and outer-only-constrained port shapes in one entity) whose `testsuite.sh` analyzes, elaborates, and simulates the design (with `--fst`, matching the report's wave-dump usage) and asserts it completes normally with no `GHDL Bug occurred`. This locks in the current, correct behavior as a regression guard.

closed 2026-08-25 by tgingold