ghdl/ghdl #2956: 5.0.1 fails to build on arm-linux-gnueabihf

doko42, 2025-03-30:
I see that ghdl builds on the ARM32 soft-float variant, but not on the hard-float variant. This used to work with 4.1

seen at
https://launchpadlibrarian.net/785418527/buildlog_ubuntu-plucky-armhf.ghdl_5.0.1+dfsg-1ubuntu0.1_BUILDING.txt.gz

[...]
elaborate and simulate hello
/usr/bin/ld: error: hello uses VFP register arguments, e~hello.o does not
/usr/bin/ld: failed to merge target specific data of file e~hello.o
/usr/bin/ld: error: hello uses VFP register arguments, /<<PKGBUILDDIR>>/testrundir/gcc/usr/lib/ghdl/gcc/vhdl/std/v87/std_standard.o does not
/usr/bin/ld: failed to merge target specific data of file /<<PKGBUILDDIR>>/testrundir/gcc/usr/lib/ghdl/gcc/vhdl/std/v87/std_standard.o
/usr/bin/ld: error: hello uses VFP register arguments, hello.o does not
/usr/bin/ld: failed to merge target specific data of file hello.o
collect2: error: ld returned 1 exit status
/<<PKGBUILDDIR>>/testrundir/gcc/usr/bin/ghdl-gcc:error: compilation error

any idea what to look at?

tgingold, 2025-03-30:
It looks like an ABI issue.  Some ABI flags are missing while compiling vhdl files.

doko42, 2025-03-30:
[ I'm not the Debian maintainer for ghdl, just looking at the packaging ]

What I can see from the build log, the ghdl build is configured with (search for "Configured with") the same flags as the Debian/Ubuntu gcc build, especially

	 --with-arch=armv7-a+fp
	 --with-float=hard
	 --with-mode=thumb
	 --build=arm-linux-gnueabihf
	 --host=arm-linux-gnueabihf
	 --target=arm-linux-gnueabihf

where would ghdl assume it's own ABI flags?