ghdl/ghdl #2957: 5.0.1 fails with the tests on s390x-linux-gnu (LLVM backend)

doko42, 2025-03-30:
ghdl 5.0.1 builds on s390x-linux-gnu, GCC and LLVM backends, GCC tests, but tests for the LLVM backend fail.

See
https://launchpadlibrarian.net/785418500/buildlog_ubuntu-plucky-s390x.ghdl_5.0.1+dfsg-1ubuntu0.1_BUILDING.txt.gz

I don't know if I already reported that, but that also was seen with 4.1.0.

tgingold, 2025-03-30:
This could be fun to investigate, but is it possible to ssh to such a machine ?

doko42, 2025-03-30:
No Ubuntu machine, although you could do experimentas within a PPA. There's limited access to the Debian porter boxes, but the GCC build farm should also have a s390x machine.

tgingold, 2025-04-02:
Unfortunately, there is no s390x machine in the GCC build farm.  qemu could be a way...

I close this issue unless there is more interest!

closed 2025-04-02 by tgingold