ghdl/ghdl #2958: GCC backend for riscv64-linux-gnu fails to build

doko42, 2025-03-30:
full build log at
https://launchpadlibrarian.net/785439128/buildlog_ubuntu-plucky-riscv64.ghdl_5.0.1+dfsg-1ubuntu0.1_BUILDING.txt.gz

[...]
cd lib/ghdl/gcc/vhdl/std/v87; \
for i in std/v87/textio.vhdl std/v87/textio-body.vhdl; do \
  cmd=" /<<PKGBUILDDIR>>/builddir/gcc/build/gcc/ghdl -a --GHDL1=/<<PKGBUILDDIR>>/builddir/gcc/build/gcc/ghdl1  --std=87 --bootstrap --work=std ../../src/$i"; \
  echo $cmd; eval $cmd || exit 1; \
done
/<<PKGBUILDDIR>>/builddir/gcc/build/gcc/ghdl -a --GHDL1=/<<PKGBUILDDIR>>/builddir/gcc/build/gcc/ghdl1 --std=87 --bootstrap --work=std ../../src/std/v87/textio.vhdl
textio.s: Assembler messages:
textio.s:3: Error: the architecture string of -march and elf architecture attributes cannot be empty
/<<PKGBUILDDIR>>/builddir/gcc/build/gcc/ghdl:error: compilation error
make[3]: *** [../../libraries/Makefile.inc:187: lib/ghdl/gcc/vhdl/std/v87/std-obj87.cf] Error 1
make[3]: Leaving directory '/<<PKGBUILDDIR>>/builddir/gcc'
make[2]: *** [Makefile:334: libs.vhdl.gcc] Error 2

the GCC backend is configured with the same configure options as the GCC build:

	 --with-arch=rv64gc
	 --with-abi=lp64d
	 --enable-checking=release
	 --build=riscv64-linux-gnu
	 --host=riscv64-linux-gnu
	 --target=riscv64-linux-gnu
	 --with-build-config=bootstrap-lto-lean
	 --enable-link-serialization=4

tgingold, 2025-04-02:
For #2956, #2957 and this #2958 issue, I think that's the same problem: ghdl directly spawn ghdl1 (which corresponds to cc1), without using the gcc front-end.  Therefore, some flags (like -mabi, -march) are not added and ghdl1 compiles for a slightly different architecture.

A few solutions: either `ghdl` can call `gcc` (probably renamed `ghdl-gcc`.  That will require some quirks during installation.
Or the user can specify flags to be added by `ghdl` when `ghdl1` is called.  This is not very user friendly at configure time.

I don't plan to work on this soon, I plan to phase out the gcc backend.  But I will keep this issue open.

doko42, 2025-04-03:
The first suggested solution sounds plausible. where would the calling location be in ghdl? I don't think I would need the ghdl-gcc name, if I know about the called gcc, ensured by package dependencies.

tgingold, 2025-04-04:
I am not sure to understand your reply.  Why do you mean by `calling location be in ghdl` ?
`ghdl1` is spawned by `src/ghdldrv/ghdldrv.adb` and looked on the PATH.
If it spawned the standard `gcc`, I fear it won't work as `gcc` doesn't know about vhdl files.

We can try to spawn a specific `gcc` (or a `ghdl-gcc`).  But then we also need to handle the location of where `ghdl1` is placed.

Not very difficult, but not trivial too.

closed 2025-04-18 by tgingold