ghdl/ghdl #2961: Synthesis of ROM reverses memory contents and adds unneeded subtractor on address

marzoul, 2025-03-31:
**Description**
When synthesizing a read-only memory, GHDL emits reverses memory contents. There is an additional subtractor on the address path, which should not have been present in the first place. This adds burden on Synthesis tools perform the opposite operation and remove the subtractor, and adds uncertainty on the efficiency of the synthesized netlist.

**Expected behaviour**
GHDL should not transform the design and emit the ROM contents as provided by the user.

**How to reproduce?**

```vhd :file: top.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity top is
	port (
		clk    : in  std_logic;
		rddata : out std_logic_vector(7 downto 0)
	);
end top;

architecture synth of top is

	constant MEM_SIZE : natural := 16;

	type rom_type is array (0 to MEM_SIZE-1) of integer range 0 to 255;

	constant rom : rom_type := (
		0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15
	);

	signal read_index : integer range 0 to 255 := 0;

begin

	rddata <= std_logic_vector(to_unsigned(rom(read_index), 8));

	process (CLK)
	begin
		if rising_edge(CLK) then

			read_index <= read_index + 1;

		end if;
	end process;

end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

Synthesis result is :
```verilog
module top
  (input  clk,
   output [7:0] rddata);
  reg [7:0] read_index;
  wire [3:0] n2_o;
  wire [3:0] n4_o;
  wire [31:0] n12_o;
  wire [31:0] n14_o;
  wire [7:0] n15_o;
  reg [7:0] n18_q;
  wire [7:0] n20_data; // mem_rd
  assign rddata = n20_data; //(module output)
  /* top.vhd:23:16  */
  always @*
    read_index = n18_q; // (isignal)
  initial
    read_index = 8'b00000000;
  /* top.vhd:27:52  */
  assign n2_o = read_index[3:0];  // trunc
  /* top.vhd:27:52  */
  assign n4_o = 4'b1111 - n2_o;
  /* top.vhd:33:50  */
  assign n12_o = {24'b0, read_index};  //  uext
  /* top.vhd:33:50  */
  assign n14_o = n12_o + 32'b00000000000000000000000000000001;
  /* top.vhd:33:39  */
  assign n15_o = n14_o[7:0];  // trunc
  /* top.vhd:31:17  */
  always @(posedge clk)
    n18_q <= n15_o;
  initial
    n18_q = 8'b00000000;
  /* top.vhd:9:17  */
  reg [7:0] n19[15:0] ; // memory
  initial begin
    n19[15] = 8'b00000000;
    n19[14] = 8'b00000001;
    n19[13] = 8'b00000010;
    n19[12] = 8'b00000011;
    n19[11] = 8'b00000100;
    n19[10] = 8'b00000101;
    n19[9] = 8'b00000110;
    n19[8] = 8'b00000111;
    n19[7] = 8'b00001000;
    n19[6] = 8'b00001001;
    n19[5] = 8'b00001010;
    n19[4] = 8'b00001011;
    n19[3] = 8'b00001100;
    n19[2] = 8'b00001101;
    n19[1] = 8'b00001110;
    n19[0] = 8'b00001111;
    end
  assign n20_data = n19[n4_o];
  /* top.vhd:27:52  */
endmodule
```

**Context**
Please, provide the following information:

- OS: Archlinux
- Origin: Built from fresh sources, AUR repository

**Additional context**

Potentially linked to other issue https://github.com/ghdl/ghdl/issues/2930 
The run time and memory footprint of Yosys may be affected by memory implementation optimizations, compared to equivalent Verilog input that does not reverse the memory contents.

tgingold, 2025-04-02:
A pass could be added to reverse memories in that case.

But contrary to verilog (I think), there is no a single direction for an index: it can be changed and viewed differently.

marzoul, 2025-04-02:
But what makes ghdl "reverse" the memory / array contents in the first place, compared to user-provided input ?
Il the vast majority of situations this represents precisely the desired implementation.

tgingold, 2025-04-03:
It doesn't reverse the memory, it just fills it from left to right.

marzoul, 2025-04-05:
Well, in the input VHDL code, the array range is `0 to MEM_SIZE-1` and the init is provided in that same order, from left to right, so from 0 to `MEM_SIZE-1` too. The order of values would be the same in verilog too. So it is very much expected to get the same assignment of cell index 0 in synthesis results.

I can propose 2 reasons for ghdl to keep the order of the contents of the synthesized RAM or ROM :
- Validation tests may check the raw memory contents after some design execution/simulation,
- in FPGA designs it is common practice to create one implemented bitstream, and patch/replace the initial memory contents for different execution parameters, or with a different firmware/microcode, etc.

To my eyes it might only makes sense to produce "reversed" memory contents if the array range is declared "MEM_SIZE-1 downto 0". Ideally under a switch to enable it, or in presence of a user attribute, because the index 0 if the raw memory/array would still contain different values before/after synthesis.

tgingold, 2025-04-06:
I am not against keeping the same order as the index, but that's not that simple.  In particular this shouldn't be done for vectors (which are arrays).

commit f42a841a (2025-05-12): netlists-memories.adb: reverse memory content if needed

Fix #2961
closed 2025-05-12 by tgingold