ghdl/ghdl #2964: LCOV reporting branch coverage for non-branch lines in VHDL code compile with GHDL - gcc backend

aaghmour, 2025-04-02:
**Description**
I'm working with GHDL (GHDL - v6.0.0-dev - Ubuntu 24.04 (x86-64, LTS) - gcc backend) and LCOV to analyze code coverage for VHDL projects, but I'm encountering unexpected behavior in the branch coverage reporting. The Issue: LCOV is showing branch coverage metrics for lines of code that don't appear to contain actual branches. Here's an example (lcov report) from a simple adder component:
```
Branch data     Line data    Source code
       1                 :          13 : entity adder is
       2                 :             :   -- `i0`, `i1` and the carry-in `ci` are inputs of the adder.
       3                 :             :   -- `s` is the sum output, `co` is the carry-out.
       4                 :             :   port (i0, i1 : in bit; ci : in bit; s : out bit; co : out bit);
       5                 :             : end adder;
       6                 :             : 
       7                 :           2 : architecture rtl of adder is
       8                 :             : begin
       9                 :             :    --  This full-adder architecture contains two concurrent assignment.
      10                 :             :    --  Compute the sum.
      11   [ -  +  +  + ]:          17 :    s <= i0 xor i1 xor ci;
      12                 :             :    --  Compute the carry.
      13   [ +  +  +  +  :          10 :    co <= (i0 and i1) or (i0 and ci) or (i1 and ci);
          +  +  +  +  +  
             +  -  +  +  
                      + ]
      14                 :             : end rtl;
      15                 :             :
```

Note that line 11 (s <= i0 xor i1 xor ci;) shows branch coverage data ([ - + + + ]) despite being a simple XOR operation with no explicit branches. I would expect branch coverage to only appear for control flow statements like if, case, etc.

Additional Output from gcov Here's the raw .gcov output from running gcov -b -c -s . adder.vhd, which may help clarify what's happening under the hood:

```
-:    0:Source:/home/aaghmour/ghdl-coverage/projects/adder/adder.vhd
        -:    0:Graph:adder.gcno
        -:    0:Data:adder.gcda
        -:    0:Runs:1
       13:    1:entity adder is
        -:    2:  -- `i0`, `i1` and the carry-in `ci` are inputs of the adder.
        -:    3:  -- `s` is the sum output, `co` is the carry-out.
        -:    4:  port (i0, i1 : in bit; ci : in bit; s : out bit; co : out bit);
        -:    5:end adder;
        -:    6:
        2:    7:architecture rtl of adder is
        -:    8:begin
        -:    9:   --  This full-adder architecture contains two concurrent assignment.
        -:   10:   --  Compute the sum.
      17*:   11:   s <= i0 xor i1 xor ci;
        -:   12:   --  Compute the carry.
      10*:   13:   co <= (i0 and i1) or (i0 and ci) or (i1 and ci);
call    0 returned 1
branch  1 taken 4 (fallthrough)
branch  2 taken 4
branch  3 taken 6 (fallthrough)
branch  4 taken 2
branch  5 taken 2 (fallthrough)
branch  6 taken 4
branch  7 taken 5 (fallthrough)
branch  8 taken 3
branch  9 taken 2 (fallthrough)
branch 10 taken 3
branch 11 taken 0 (fallthrough)
branch 12 taken 8
branch 13 taken 3 (fallthrough)
branch 14 taken 5
call   15 returned 3
------------------
work__adder__ARCH__rtl__STMT_ELAB:
function work__adder__ARCH__rtl__STMT_ELAB called 1 returned 100% blocks executed 100%
        1:    1:entity adder is
        -:    2:  -- `i0`, `i1` and the carry-in `ci` are inputs of the adder.
        -:    3:  -- `s` is the sum output, `co` is the carry-out.
        -:    4:  port (i0, i1 : in bit; ci : in bit; s : out bit; co : out bit);
        -:    5:end adder;
[... 158 lines cut ...]
```
1. create a build folder
2. compile & run testbench then generate coverage repport:
cd build/ && ghdl -a --workdir=. -Wc,-fprofile-arcs -Wc,-ftest-coverage -fno-exceptions -fno-inline ../projects/adder/adder.vhd
cd build/ && ghdl -a --workdir=.  ../projects/adder/adder_tb.vhd
cd build/ && ghdl -e --workdir=. -Wl,-lgcov adder_tb
cd build/ && ghdl -r --workdir=. adder_tb
cd build/ && gcov -b -c  -s . adder.vhd

The full project code is available at: https://github.com/huettern/ghdl-coverage

**Context**
Please, provide the following information:

- OS: Ubuntu 24.04.2 LTS
- Origin:
  - [ ] Package manager: `version`
  - [x] Released binaries: `https://github.com/ghdl/ghdl/releases/download/v5.0.1/ghdl-gcc-5.0.1-ubuntu24.04-x86_64.tar.gz`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

**Additional context**
screenshot of the generate coverage report
![Image](https://github.com/user-attachments/assets/5154c022-0b11-4697-b5b1-2fd5cddc7bba)

tgingold, 2025-04-03:
`or` is a short-circuit operator, so there are implicit conditions.
Due to various inserted checks, gcov condition coverage does not correspond to what the user could see.
You can use the coverage provided by the mcode backend (`--coverage`), or if you use the gcc backend, use only statement coverage.

closed 2025-04-03 by tgingold

aaghmour, 2025-04-04:
Hello,

Could anyone guide me on how to use the --coverage flag properly with the MCode backend to collect coverage data?

Thank you in advance!

I've successfully compiled GHDL with the MCode backend and installed it. When I use the --coverage flag to compile and elaborate, the tests run fine, but I’m having trouble collecting coverage data specifically with the MCode backend.

Here is what I did to compile and elaborate with ghdl mcode:
```
cd build/ && ghdl -a --workdir=. --coverage ../projects/adder/adder.vhd
cd build/ && ghdl -a --workdir=.  ../projects/adder/adder_tb.vhd
cd build/ && ghdl -e --workdir=. --coverage adder_tb
cd build/ && ghdl -r --coverage --workdir=. adder_tb
../projects/adder/adder_tb.vhd:53:7:@8ns:(assertion note): end of test
```

In the past, when using the GCC backend, I was able to collect coverage data with the following flags and using gcov & lcov:

`-Wc,-fprofile-arcs -Wc,-ftest-coverage`

However, I’m unsure how to achieve the same result with the MCode backend.
Thank you

tgingold, 2025-04-05:
There is an example in the testsuite:
https://github.com/ghdl/ghdl/blob/master/testsuite/sanity/008coverage/testsuite.sh

You need to run with `--coverage` (as you did).
This should generate a `.json` file.
You can then gather them (if you do multiple runs) and generate outputs in different formats using the `ghdl coverage` command.