ghdl/ghdl #2966: Synthesis declares duplicate signal

JJ-Gaisler, 2025-04-04:
**Description**
When synthesizing a GRLIB DMA entity GHDL produces duplicate declarations of signals leading to non-compilable code,

**Expected behaviour**
Separate identifiers would be used

**How to reproduce?**
While not a minimal example it is possible to reproduce the problem by doing the following:

1. Download [grlib](https://download.gaisler.com/products/GRLIB/bin/grlib-gpl-2025.1-b4296.tar.gz)
2. Navigate to designs/noelv-generic/
3. Replace file rtl/noelvmp.vhd with file attached in issue (.txt extension for github to accept it)
4. run: make ghdl
5. `ghdl --synth --out=verilog -fexplicit --ieee=synopsys --mb-comments --warn-no-binding --workdir=gnu/work --work=work  `cat ghdl.path` --warn-no-hide --warn-no-specs --warn-no-port gaisler.grdmac2 > out.v`
6. Check verilog for duplicated identifiers: `fifo_rusedw, ctrl_rst` or by running `verilator --lint-only out.v`

```
%Error: out.v:14062:14: Duplicate declaration of signal: 'fifo_rusedw'
14062 |   wire [3:0] fifo_rusedw;
      |              ^~~~~~~~~~~
        out.v:13653:14: ... Location of original declaration
13653 |   wire [3:0] fifo_rusedw;
      |              ^~~~~~~~~~~
%Error: out.v:1020:11: Duplicate declaration of signal: 'ctrl_rst'
 1020 |    output ctrl_rst,
      |           ^~~~~~~~
        out.v:907:11: ... Location of original declaration
  907 |    input  ctrl_rst,
      |           ^~~~~~~~
%Error: out.v:1020:11: Duplicate declaration of port: 'ctrl_rst'
 1020 |    output ctrl_rst,
      |           ^~~~~~~~
        out.v:907:11: ... Location of original declaration
  907 |    input  ctrl_rst,
      |           ^~~~~~~~
%Error: out.v:14839:6: Duplicate pin connection: 'ctrl_rst'
14839 |     .ctrl_rst(ctrl_ctrl_rst),
      |      ^~~~~~~~
        out.v:14726:6: ... Location of original pin connection
14726 |     .ctrl_rst(n294_o),
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [x] Built from sources: `f52e15784a373d39d94444fe86a372eeae7f3f21`

[noelvmp.txt](https://github.com/user-attachments/files/19599861/noelvmp.txt)

tgingold, 2025-04-06:
It is 'just' because an instance named 'fifo' has a port named 'rusedw' (cf frdmac2.vhd line 399).
If you rename the instance to 'sfifo', it shouldn't fail.

I have to investigate why this happens...

tgingold, 2025-04-07:
`ctrl_rst` is a little bit different: the clash appears in `grdmac2_ctrl` as there is both `ctrl_rst` port and `ctrl.rst` port.

closed 2025-04-18 by tgingold

JJ-Gaisler, 2025-04-20:
Thanks for the update! It looks like the duplicated identifiers are no longer a problem. However, the update brought other naming issues with it. This is from the same example as in the initial reproducer. Rebuilt GHDL to `b6.0.0.r88.g4c416ec36`

Output from verilator linting (subset of all errors)

```
%Error: out.v:13411:20: syntax error, unexpected IDENTIFIER, expecting ',' or ';'
13411 |   wire [127:0] b2m_\buf_out[wdata] ;
      |                    ^~~~~~~~~~~~~~~
%Error: out.v:13450:20: syntax error, unexpected IDENTIFIER, expecting ',' or ';'
13450 |   wire [31:0] ctrl_\curr_desc_out[dbg_ctrl] ;
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~
%Error: out.v:13451:20: syntax error, unexpected IDENTIFIER, expecting ',' or ';'
13451 |   wire [31:0] ctrl_\curr_desc_out[dbg_nxt] ;
      |                    ^~~~~~~~~~~~~~~~~~~~~~~
```

JJ-Gaisler, 2025-04-22:
@tgingold It looks like line https://github.com/ghdl/ghdl/blob/a4d334fb72abb78bf25f9d1eb3a2f25eadf61a97/src/synth/netlists-disp_verilog.adb#L68 is the line responsible. I have still not quite understood when to escape. Simply removing the escaping causes issues with names such as class so escaping makes sense. However I don't see the need to escape in this particular instance. The name seems to be created from the instantiation label and the port name.

JJ-Gaisler, 2025-04-28:
Again I really appreciate the great support and effort @tgingold ! Your last commit looks to have solved all of the escaping issues, therefore I proceeded to synthesize the entity that instantiates the dma unit. When doing so I get a new duplicated declaration error.
```
%Error: noelv.v:540827:24: Duplicate declaration of instance: 'n1_encoder'
540827 |   aplic_encoder_31_5_8 n1_encoder (
       |                        ^~~~~~~~~~
        noelv.v:540813:24: ... Location of original declaration
540813 |   aplic_encoder_31_5_8 n1_encoder (

```
The following can be appended to the makefile, running make noelv.v to reproduce the issue.

```make
PREFIX_GENERICS = \
-gfabtech=0 \
-gmemtech=0 \
-gmularch=0 \
-gncpu=1 \
-gnextmst=0 \
-gnextslv=1 \
-gnextapb=0 \
-gndbgmst=1 \
-gnintdom=0 \
-gneiid=63 \
-gcached=0 \
-gwbmask=16\#50F\# \
-gbusw=128 \
-gcmemconf=0 \
-grfconf=1 \
-gfpuconf=0 \
-gtcmconf=0 \
-gmulconf=0 \
-gdisas=0 \
-gahbtrace=0 \
-gcfg=0 \
-gdevid=0 \
-gnodbus=0 \
-gtrace=0 \
-gscantest=0

TOP = gaisler.noelvsys
VERILOG_OUT=noelv.v
ifeq ($(YOSYS), 1)

YOSYS_OPTS= proc; peepopt; fsm; opt;  memory; opt; clean; check; write_verilog -sv $(VERILOG_OUT)

$(VERILOG_OUT): ghdl
         yosys -m ghdl -p 'ghdl --no-formal -fexplicit --ieee=synopsys --mb-comments --warn-no-binding --warn-no-hide $(PREFIX_GENERICS) $(shell cat ghdl.path) -O0 --workdir=gnu/work --work=work $(TOP); $(YOSYS_OPTS)'
else
$(VERILOG_OUT): ghdl
        $(GHDL) --synth --out=verilog --no-formal -fexplicit --ieee=synopsys --mb-comments --warn-no-binding --workdir=gnu/work $(PREFIX_GENERICS) --work=work  $(shell cat ghdl.path) --warn-no-hide --warn-no-specs --warn-no-port $(TOP) > $(VERILOG_OUT)
        verilator --lint-only -Wno-unoptflat noelv.v
endif
```

If the unit that produces the duplicate identifier is disabled I can synthesize the entire processor with one warning.
This can be replicated by setting AIA_SUPPORT to 0 in $GRLIB/lib/gaisler/noelv/pkg/noelv.vhd
```
%Warning-WIDTHEXPAND: noelv.v:199980:38: Operator ADD expects 32 or 13 bits on the LHS, but LHS's VARREF 'n316068.o' generates 9 bits.
199980 |   assign \n302645.o  = r[\n316068.o  + 8033 -: 32];
```

tgingold, 2025-04-28:
The duplicate identifier should be fixed.
I still have to check the yosys plugin and the reason of the widthexpand warning

tgingold, 2025-05-01:
I don't think the generated code is incorrect.  Only a part of the whole memory is read, so it is OK to have less bits.

I don't plan to change the code, unless we want to get rid of that warning.

JJ-Gaisler, 2025-05-05:
> I don't think the generated code is incorrect. Only a part of the whole memory is read, so it is OK to have less bits.
> 
> I don't plan to change the code, unless we want to get rid of that warning.

Okay that is fine with me! The code generated by commit 832fcbc contains a few new naming errors but I like the direction that synth is heading. It is has become significantly easier to track what is going on now when simulating the "netlist".

```
%Error: noelv.v:547964:5: syntax error, unexpected '.', expecting IDENTIFIER or randomize                                                                                                                                                                                                                                    
547964 |     .\pbo_pbo[data]\(\pbgen_x_n1_pb0.pbo_pbo[data] ));                                                                                                                                                                                                                                                              
       |     ^     
```

Steps to reproduce is the same as before, synthesizing noelv.

tgingold, 2025-05-05:
Oops, was a stupid mistake.  Now fixed.

commit a4d334fb (2025-04-18): synth: use extended/escaped names for records

Fix #2966