ghdl/ghdl #2970: GHW: array indices are reversed

JakubFranek, 2025-04-12:
**Description**

**TL;DR:** When plotting a GHW wave in Surfer, array indices appear reversed (i.e. member at index 0 behaves as member at index N-1 should and vice versa). When connecting to the array members with a standard non-array signal and plotting the signal, the values are indexed correctly. Maybe GHW file is incorrect?

I have a BCD counter, which is counting using the following array type:
`type t_bcd_array_debug is array (natural range <>) of unsigned(3 downto 0);`

The BCD counter has a generic number of digits and it is designed so that digit at index 0 is the least significant digit. It's output is:
`o_decimals : out   t_bcd_array_debug(DECIMALS - 1 downto 0)`

The testbench is VUnit driven. I have auxiliary signals dec0, dec1 and dec2 which connect to the output array members (in my case DIGITS = 3).

```
inst_uut : entity work.bcd_counter_debug
        generic map (
            DECIMALS => DIGITS
        )
        port map (
            i_clk      => clk,
            o_decimals => decimals
        );    

    dec0 <= decimals(0);
    dec1 <= decimals(1);
    dec2 <= decimals(2);
```

When I plot dec0, dec1 and dec2 in Surfer, I get expected result (dec0 is the least significant digit). But when I plot the array members themselves, I get reversed order of values (decimals(2) is the least significant digit):

[image]

I have two suspects: either the indices are reversed in the GHW file itself (my main suspect), or the Surfer waveform viewer is bugged and reverses the indices. I don't know how to read the GHW file to check whether the indices within the file are correct or not though.

**How to reproduce?**
The VHDL files for the counter, its testbench and the package are pasted below.
They are also zipped along with the GHW file in this ZIP: [bcd_counter_debug.zip](https://github.com/user-attachments/files/19719187/bcd_counter_debug.zip)

GHDL version 5.0.1
OS Windows 11
Surfer version 0.3.0

bcd_counter_debug.vhd:
```
library ieee;
    use ieee.std_logic_1164.all;
    use work.bcd_record_package_debug.all;
    use ieee.numeric_std.all;

entity bcd_counter_debug is
    generic (
        DECIMALS   : natural range 1 to natural'high := 4;
        SIMULATION : boolean                         := true
    );
    port (
        i_clk      : in    std_logic;
        o_decimals : out   t_bcd_array_debug(DECIMALS - 1 downto 0)
    );
end entity bcd_counter_debug;

architecture rtl of bcd_counter_debug is

    signal q_counter : t_bcd_array_debug(DECIMALS - 1 downto 0) := (others => "0000");

begin

    proc_clk : process (i_clk) is

        variable v_carry : boolean := true;

    begin

        if rising_edge(i_clk) then
            v_carry := true;

            for i in 0 to DECIMALS - 1 loop

                if (v_carry) then
                    if (q_counter(i) = "1001") then
                        q_counter(i) <= (others => '0');
                        v_carry      := true;
                    else
                        q_counter(i) <= q_counter(i) + 1;
                        v_carry      := false;
                    end if;
                end if;

            end loop;

        end if;

    end process proc_clk;

    o_decimals <= q_counter;

end architecture rtl;
```

bcd_counter_debug_tb.vhd:
```
library ieee;
    use ieee.std_logic_1164.all;
    use work.bcd_record_package_debug.all;
    use ieee.numeric_std.all;

library vunit_lib;
    context vunit_lib.vunit_context;

entity bcd_counter_debug_tb is
    generic (
        RUNNER_CFG : string;
        DIGITS     : natural := 3
    );
end entity bcd_counter_debug_tb;

architecture tb of bcd_counter_debug_tb is

    signal   clk        : std_logic := '0';
    signal   decimals   : t_bcd_array_debug(DIGITS - 1 downto 0);
    constant CLK_PERIOD : time      := 20 ns;

    signal dec0 : unsigned (3 downto 0);
    signal dec1 : unsigned (3 downto 0);
    signal dec2 : unsigned (3 downto 0);

begin

    inst_uut : entity work.bcd_counter_debug
        generic map (
            DECIMALS => DIGITS, SIMULATION => true
        )
        port map (
            i_clk      => clk,
            o_decimals => decimals
        );

    clk <= not clk after (CLK_PERIOD / 2);

    dec0 <= decimals(0);
    dec1 <= decimals(1);
    dec2 <= decimals(2);

    proc_test_runner : process is
    begin

        test_runner_setup(runner, RUNNER_CFG);

        show(get_logger(default_checker), display_handler, pass);

        wait until rising_edge(clk);

        wait for 150 * CLK_PERIOD;

        test_runner_cleanup(runner); -- Simulation ends here

    end process proc_test_runner;

end architecture tb;

```

bcd_record_package_debug.vhd
```
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;

package bcd_record_package_debug is

    type t_bcd_array_debug is array (natural range <>) of unsigned(3 downto 0);

end package bcd_record_package_debug;
```

GHW dump produces the following:

```
$ ghwdump -l -t -v simulation/vunit_out/test_output/lib.bcd_counter_tb.count_up_064dda88058e05b00f313eb7872c0eff9807fc64/ghdl/wave.ghw 
Section STR
Number of strings: 90
String table size: 886
Section TYP
type std_ulogic is ('U', 'X', '0', '1', 'Z', 'W', 'L', 'H', '-');
subtype std_logic is std_ulogic range 'U' to '-';
type integer is range <>;
subtype natural is integer range 0 to 2147483647;
type std_ulogic_vector is array (natural range <>) of std_ulogic;
subtype runner_sync_t is std_ulogic_vector (0 to 20);
type unresolved_unsigned is array (natural range <>) of std_ulogic;
subtype <anon> is unresolved_unsigned (3 downto 0);
type t_bcd_array is array (natural range <>) of unresolved_unsigned (3 downto 0);
subtype <anon> is t_bcd_array (2 downto 0);
subtype <anon> is unresolved_unsigned (3 downto 0);
subtype <anon> is unresolved_unsigned (3 downto 0);
subtype <anon> is unresolved_unsigned (3 downto 0);
subtype <anon> is t_bcd_array (2 downto 0);
subtype <anon> is t_bcd_array (2 downto 0);
Section WKT
std_ulogic: wkt=3
Section HIE
9 scopes, 19 signals, 64 signal elements
Section EOH
Section SNP
Section CYC
Section DIR
Directory (8 entries):
 STR at 16
 TYP at 770
 WKT at 876
 HIE at 887
 EOH at 1206
 SNP at 1210
 CYC at 1294
 DIR at 5250
Section TAI
Tailer: directory at 5250
eof of file
```

marzoul, 2025-04-12:
This looks similar to issue https://github.com/ghdl/ghdl/issues/2961
The GHDL tools may use an internal mapping of addresses that does not mirror "user" indices in array.

diogratia, 2025-04-12:
A several year old gtkwave in MacOS (on an M1):

```none
%: cp wave.ghw bcd_counter_debug_tb.ghw
%: gtkwave bcd*.ghw

GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

[0] start time.
[3000000000] end time.
````
(The rename was to use a gtkw file consistent with a Makefile target while preserving your original ghw wavedump file.)

[image]

There is no apparent silo to name transposition visible here. As a first guess I'd say there's a Surfer issue.

The surfer_macos_64 binary from 8/29/24 shows

[image]

them reversed as you show.

Recreating your waveform from source files (after removing vunit references and adding std.env.stop to the proc_test_runner process in the testbench as a last statement):

[image]

duplicates wave.ghw (renamed):

A conclusion you could make is that surfer didn't faithfully duplicate the functionality of the gtkwave.

Paebbels, 2025-04-13:
As far as I know, Surfer doesn't use libghw, but has it's own implementation of the GHW format. Don't ask me why they do it, but GHW isn't officially documented, thus binary compatibility and changes to the GHW format will always be a problem to Surfer.

tgingold, 2025-04-13:
Yes, this looks like a Surfer issue.

closed 2025-04-13 by tgingold

JakubFranek, 2025-04-13:
Thanks a lot to everyone, especially @diogratia, for debugging this so thoroughly and quickly. 

In case someone wants to track this the bug further, I have created an [issue over at the Surfer GitLab](https://gitlab.com/surfer-project/surfer/-/issues/401).