ghdl/ghdl #2971: ghdl-llvm-jit-5.0.1-macos15-aarch64.tar.gz binaries don't analyse to object files

sebeaumont, 2025-04-12:
**Description**
Having downloaded the above binary package for `ghdl` all seems to work ok except that analysis does not produce any objects and thus cannot build simulation program even tho' `ghdl` claims to have been built with llvm backend. 

**Expected behaviour**
I'd like some object files and I don't get any. Or any errors or warnings.

**How to reproduce?**
`ghdl -a foo.vhd` 

**Context**
Please, provide the following information:

- OS: macOS Sequoia 15.4 (aarch64) M1
- Origin:
  - [ ] Package manager: `version`
  - [x] Released binaries: `ghdl-llvm-jit-5.0.1-macos15-aarch64.tar.gz`
  - [ ] Built from sources: `commit SHA`

```
GHDL 5.0.1 (4.1.0.r602.g37ad91899) [Dunoon edition]
 Compiled with GNAT Version: 14.2.0
 static elaboration, LLVM JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2025 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tgingold, 2025-04-12:
It's the jit version.  No object files are created.

closed 2025-04-12 by tgingold

tgingold, 2025-04-12:
The command line is the same: `-a` to analyze, `-r` to run.

sebeaumont, 2025-04-12:
In the end I built myself a full llvm version with the `alire gnat_native` compiler and also enabled synthesis with the shared library for Yosys so I can probably do the works now (subject to building the plugin next) tho' I'm only worried about simulation on this box. I have a lab machine with old fashioned x86 chipset running Fedora 41 for talking to hardware. 
```
GHDL 6.0.0-dev (b6.0.0.r66.g32eea8203) [Dunoon edition]
 Compiled with GNAT Version: 14.2.0
 llvm 14.0.6 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2025 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
``` 
I used llvm14 as that was the latest recommendation on the GitHub build notes.

Paebbels, 2025-04-13:
The releases page offers a normal LLVM version and the LLVM + JIT version for macOS. GHDL is offered as cask through homebrew.