ghdl/ghdl #2972: Failing pipeline due to Python Packaging Authority

Paebbels, 2025-04-13:
The pipeline are failing to install Python packages, because PyPA (Python Packaging Authority) changed how to normalize package names. This stupid change is rolled out in different speed to our various platforms. Thus the hit faster on rolling releases like MSYS2 and might reach very late at Ubuntu or macOS. Moreover, we might have fun in Ubuntu because some GitHub CI images will use the old normalization and others will use the new normalization.

Even more crazy, why I have no appreciation, Python is case sensitive and allows for mixed case package names. Now we have the situation to deliver Python packages in an installation file with a different name ... In addition, the package name is still mixed case, just the installer file is all-lower case.

Paebbels, 2025-04-13:
@tgingold do you fix this in one of you next pushes or should I prepare a small PR?

tgingold, 2025-04-13:
Please, do the PR.

Paebbels, 2025-04-13:
I'm on it. I got many private notes and other issues here on GitHub about this issue.

commit c88b1e49 (2025-04-13): Adjust wheel file name spelling because PyPA changes the package name normalization rules.

Related to 5c97ac274926173727a6eaa3cdc0a5ca6b0999a9.

Fixes #2972.
Fixes #2973.
closed 2025-04-13 by tgingold