ghdl/ghdl #2973: [setup-ghdl] "nightly" tag gone?

stnolting, 2025-04-13:
I am using the CI action to get GHDL:

```
- name: '📦 Install GHDL'
  uses: ghdl/setup-ghdl@v1
  with:
    version: nightly
    backend: mcode
```

However, this fails.

[image]

I'm no expert, but it seems like the "nightly" tag is gone. Hardcoding the latest version works:

- name: '📦 Install GHDL'
  uses: ghdl/setup-ghdl@v1
  with:
    version: v5.0.1
    backend: mcode
```

Paebbels, 2025-04-13:
This is related to #2972. I pushed/moved the `nightly` tag. I'll try to provide a PR today (stupid PyPA ...).

Please note due to another incompatibility of MSYS2 and LLVM, not all GHDL variants are supported by `setup-ghdl` (we needed to delete MSYS2 + LLVM combination).

Paebbels, 2025-04-13:
@stnolting, @jimlewis

A patch is running at Paebbels/GHDL (35 minutes). If this is ok, we can merge (+ review + 35 minutes). Then we can move the nightly tag (+ trigger time + 35 minutes).

closed 2025-04-13 by tgingold

Paebbels, 2025-04-13:
Nightly has been moved. (35 minutes)

stnolting, 2025-04-13:
That was fast! Thank you very much!

Paebbels, 2025-04-13:
> That was fast! Thank you very much!

Because we had the same issue already in https://github.com/ghdl/ghdl/commit/5c97ac274926173727a6eaa3cdc0a5ca6b0999a9 for Ubuntu :).

commit c88b1e49 (2025-04-13): Adjust wheel file name spelling because PyPA changes the package name normalization rules.

Related to 5c97ac274926173727a6eaa3cdc0a5ca6b0999a9.

Fixes #2972.
Fixes #2973.