ghdl/ghdl #2976: [synthesis] issue with latest nightly version

stnolting, 2025-04-14:
I am encountering a problem with the latest `nightly` GHDL version.

I am using GHDL's synthesis feature to convert a soft-core processor to Verilog. For this I use the GHDL GI action.

Using `version: nightly` produces Verilog code that does not work (https://github.com/stnolting/neorv32-verilog/actions/runs/14437818241; some UART output that is used for verification is broken). In contrast, using `version: v5.0.1` works fine (https://github.com/stnolting/neorv32-verilog/actions/runs/14454984910).

I can't identify the problem yet - but I will try to provide more information.
Have there been any significant changes in synthesis from `v5.0.1` to the current `nightly` version?

Paebbels, 2025-04-14:
Can you maybe name a CI run date where nightly was working in your pipeline? Then we could reduce the range of changes between 5.0.1 and now even further.

stnolting, 2025-04-15:
This one works (generated Verilog works as expected):

```
 Runner OS:     Ubuntu 24.04 (x86-64)
GHDL Backend:  mcode
GHDL Version:  5.0.1 (Git: 37ad91899ea3f311423eb91e34f42c0ae1948f79)
```

This one does not (generated Verilog behaves strange):

```
Runner OS:     Ubuntu 24.04 (x86-64)
GHDL Backend:  mcode
GHDL Version:  6.0.0-dev (Git: 41518b39f2896fdf4bc89225c5c5ed89eb1e8557)
```

JJ-Gaisler, 2025-04-15:
I can also see similar issues with the NOEL-V processor. Before commit: d7f9b8671b8e4c8ca56d10c7cf7319ef56dc66a8
Things were not working at all. After that update things worked better but faulty behavior was still encountered after simulating the synthesized verilog for a while.

tgingold, 2025-04-15:
@JJ-Gaisler , @stnolting if you have a reproducer that would be very welcome.  It's probably around the bmux gate, which is translated as `[X +: Y ]` in verilog

stnolting, 2025-04-15:
Just a question:

This is a cut-out of the broken Verilog:

```verilog
wire [1:0] buf_adr;
wire [1:0] n8546_o;
...
/* ../neorv32/rtl/core/neorv32_sysinfo.vhd:181:29  */
assign n8546_o = 2'b11 - buf_adr;
...
/* ../neorv32/rtl/core/neorv32_sysinfo.vhd:181:29  */
assign n8560_o = sysinfo[n8546_o * 32 -: 32]; //(Bmux)
```

What is the purpose of this BMUX construct? It feels like synthesis tools might have some problems converting this into a proper multiplexer (additiona and multiplication). 🤔

I never use these slices in Verilog... what will happen if `n8546_o` is zero?

marzoul, 2025-04-15:
@stnolting This was discussed (and fixed) in the context of other issue https://github.com/ghdl/ghdl/issues/2962
BMUX designates binary-selection multiplexer in ghdl internals.

stnolting, 2025-04-15:
Thabs for the info!
So this issue was fixed with https://github.com/ghdl/ghdl/commit/d7f9b8671b8e4c8ca56d10c7cf7319ef56dc66a8?

If so, I'll test that when the new nightly is available.

Paebbels, 2025-04-15:
I moved `nightly` for you both :). Needs circa 35 minutes.

marzoul, 2025-04-15:
@stnolting I see that your verilog version is reported to be slower than the verilog version. Do you know why ?
On my side I am digging these inefficiencies, and I found 2 that may hit (or have hit) you here : the mux issue (the mentioned https://github.com/ghdl/ghdl/issues/2962) and an extra operation on memory address path, something that I can also see in your code above `assign n8546_o = 2'b11 - buf_adr;` (see https://github.com/ghdl/ghdl/issues/2961).

JJ-Gaisler, 2025-04-16:
> [@JJ-Gaisler](https://github.com/JJ-Gaisler) , [@stnolting](https://github.com/stnolting) if you have a reproducer that would be very welcome. It's probably around the bmux gate, which is translated as `[X +: Y ]` in verilog

I ran equivalence checking with Yosys and right now 9230/24399 partitions are failing. I had another version of ghdl (g5af853d11) which is quite a bit older that I used as reference since I know that this version is working. I ran the test on the GRLIB ahbctrl module. It shows that the msti_hrdata output can read the wrong value which seems to at least be vaguely related to to what I was seeing during simulation ( a load instruction reading the wrong value )

Hopefully I have time to take a deeper look later today.

I have an old issue in which I did something similar: [issue](https://github.com/ghdl/ghdl/issues/2746)
Right now I set the top to ahbctrl: `ahbctrl_0_1_1_0_4095_4095_4080_4080_3_6_1_0_0_0_1_1_0_0_0_0_0_0_0_0_1_1_0_0_0_0_0_1_0_1`

stnolting, 2025-04-16:
@Paebbels Thank you so much for the quick update! With the recent `nightly` version everything works fine again. 🚀

@marzoul I am not sure where this is coming from. But I will have a look at the issues you've referenced. Thanks for that!

Paebbels, 2025-04-16:
@stnolting I'm just happy how well the pipeline works considering how many jobs and steps we have in that pipeline. It's the most complicated pipeline I ever created. AND how easy we can deliver a new release including automated notification emails, etc just by moving the nightly tag.

In case you have a Python package, I have a fully automated Python pipeline like this GHDL pipeline: https://github.com/pyTooling/sphinx-reports/actions/runs/14493469218 . I can release a Python package in a just a few minutes.

closed 2025-04-27 by tgingold