ghdl/ghdl #2981: unconstrained range issue

grzegorzbieszczad, 2025-04-23:
**Description**
GHDL rises a bug exception when the function uses record with unconstrained vector like in the example.

**Expected behaviour**
Function evaluation should probably take into account the inline record constrain/range-definition or should report Error/Warning gently if it is incompatible with VHDL version (IDK?).

**How to reproduce?**
Please run this example design:

```vhd :file: function_range.vhd
library ieee;
use ieee.std_logic_1164.all;

entity function_range is
    generic(DATA_WIDTH  : natural := 64);

end entity function_range;

architecture tb of function_range is

    type UNCONSTRAINED_RANGE_RECORD is record
        data : std_logic_vector;
    end record UNCONSTRAINED_RANGE_RECORD;

    procedure funtion_issue_with_range (signal b : inout UNCONSTRAINED_RANGE_RECORD(data(DATA_WIDTH - 1 downto 0))) is
    begin

    end procedure funtion_issue_with_range;

begin

end architecture tb;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a function_range.vhd
ghdl --elab-run --std=08 function_range
```
**Context**

- OS: Linux 6.8.0-57-generic \#59~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Mar 19 17:07:41 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `4c416ec3682f8caf391b5d87345a89e20e8e9344`

`GHDL Bug occurred`:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 6.0.0-dev (b6.0.0.r88.g4c416ec36) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/home/gb/Dokumenty/ghdl_issues/function_range/
Command line:
ghdl --elab-run --std=08 function_range
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : simul-vhdl_compile.adb:479 access check failed
******************************************************************
```

**Additional context**
-

diogratia, 2025-04-25:
The issue doesn't show up with the llvm back end but does show up with either the mcode or llvm-jit versions.

Experimenting around shows that if there are subprogram calls for the two subprograms you won't get this run-time error. If either one doesn't have a call the problem shows up. 

Also tried moving the  type declaration and subprogram declarations/definitions to a generic package where upon:

```none
%: $GHDL -a --std=08 function_range.vhdl
are_trees_equal: cannot handle IIR_KIND_RECORD_ELEMENT_CONSTRAINT (function_range.vhdl:23:52)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 6.0.0-dev (b6.0.0.r74.gd7f9b8671) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-apple-darwin21.6.0
/Users/dio_gratia/Desktop/
Command line:
./ghdl-mcode-6.0.0-dev-macos13-x86_64/bin/ghdl -a --std=08 function_range.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
******************************************************************
%: 
```

(There's some chance Tristan might tell us record constraints aren't fully implemented as yet. Line 23 is the procedure subprogram definition containing the subtype indication with the record constraint in the generic package body.)

tgingold, 2025-04-27:
I am testing a fix for this issue.
@diogratia, do you have a reproducer for your crash ?

tgingold, 2025-04-27:
@diogratia I think I have found one.

closed 2025-04-27 by tgingold

diogratia, 2025-04-28:
```none
%: $GHDL -a --std=08 function_range.vhdl
are_trees_equal: cannot handle IIR_KIND_RECORD_ELEMENT_CONSTRAINT (function_range.vhdl:23:52)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 6.0.0-dev (b6.0.0.r74.gd7f9b8671) [Dunoon edition]
Compiled with GNAT Version: 14.2.0
Target: x86_64-apple-darwin21.6.0
/Users/dio_gratia/Desktop/
Command line:
./ghdl-mcode-6.0.0-dev-macos13-x86_64/bin/ghdl -a --std=08 function_range.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
******************************************************************
%: 
```
Was produced by

```vhdl
library ieee;
use ieee.std_logic_1164.all;

package generic_range_issue is
    generic (DATA_WIDTH: natural := 64);
    type UNCONSTRAINED_RANGE_RECORD is record
        data : std_logic_vector;
    end record UNCONSTRAINED_RANGE_RECORD;
    
    procedure procedure_issue_with_range (
        signal b: inout 
            UNCONSTRAINED_RANGE_RECORD(data(DATA_WIDTH - 1 downto 0))); 

    function funtion_issue_with_range (
        signal b:
            UNCONSTRAINED_RANGE_RECORD(data(DATA_WIDTH - 1 downto 0)))
        return boolean;
    
end package;

package body generic_range_issue is
    procedure procedure_issue_with_range (
        signal b: inout UNCONSTRAINED_RANGE_RECORD(data(DATA_WIDTH - 1 downto 0))) is
    begin
    end procedure procedure_issue_with_range;

    function funtion_issue_with_range (
        signal b:
            UNCONSTRAINED_RANGE_RECORD(data(DATA_WIDTH - 1 downto 0)))
        return boolean is
    begin
        return FALSE;  -- ADDED
   end function funtion_issue_with_range;
 
end package body;

package range_issue is new generic_range_issue;
```

I haven't tried as a package instantiation as say a block declarative item, just as  a primary unit.

commit b54d2d89 (2025-04-27): elab: do not elaborated unused subprograms

Propagate use flag to instances.

Fix #2981