ghdl/ghdl #2982: GHDL Bug occurs when cocotb accesses nonexistent signal and generate is present

miklos-abdul, 2025-04-25:
**Description**
The "GHDL Bug occured" block appeared. I reduced the cause to using cocotb to set a nonexistent signal in a .vhdl file where a "generate" statement is present. Without the "generate", cocotb just throws an error as expected.

**How to reproduce?**
Install cocotb and run 
```
python makefile.py
```

```vhd :file: nfmac.vhdl
library IEEE;
use IEEE.std_logic_1164.all;
use ieee.numeric_std.all;
entity nfmac is generic(
    variable_name            :integer:=0
    );
port(          
    port_name:		    in std_logic         
    );
end nfmac;
architecture rtl of nfmac is
begin    
    rxfifogen:if (variable_name = 1) generate
    else generate  
    end generate rxfifogen;
end rtl; 
```

```py :file: mac_dev_tb.py
import cocotb;
async def fun(dut):
    dut.rx_dcm_locked.value=0;
@cocotb.test()
async def axis2axis_test(dut):
    await fun(dut);
```

```py :file: makefile.py
def doVHDL():
    MSRCS=["../src/nfmac.vhdl"];

    runner = get_runner('ghdl');
    runner.build(sources=MSRCS,hdl_toplevel='nfmac',includes=[VR],build_args=["--std=08"], always=True);
    runner.test(hdl_toplevel='nfmac',waves=True, test_args=["--std=08"],test_module='mac_dev_tb' );
```

**Context**
Please, provide the following information:

- OS:Ubuntu 24.04.2 LTS
- Origin:
  - [ x] Package manager: `version`
  - [ x] Released binaries: `[tarball_url](https://www.cyberciti.biz/faq/how-to-check-os-version-in-linux-command-line/)`
  - [ ] Built from sources: `commit SHA`
This bug appeared on both tarball and the apt-get version.

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.1 (4.1.0.r602.g37ad91899) [Dunoon edition]
Compiled with GNAT Version: 13.3.0
Target: x86_64-linux-gnu
/home/dgruncl/Documents/xv/devkit-fpga/ip/axi2phy/coco/sim_build/
Command line:
ghdl -r --work=top --std=08 nfmac --vpi=/home/dgruncl/Documents/libs/venv1/lib/python3.12/site-packages/cocotb/libs/libcocotbvpi_ghdl.so
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-avhpi.adb:593 access check failed
******************************************************************
```

miklos-abdul, 2025-04-25:
Addendum: If you enable waveforms using "plusargs=["--wave=vhdl.ghw"]", then the same bug occurs even if you access an existing port. 
Which means that the generate statement must be conditional, not a for loop, and it must be in top level, otherwise the would occur in another project that I have here.

miklos-abdul, 2025-04-25:
One more addition: Looking through older bugs, I found out that the variant of the bug with waveforms enabled is fixed by the solution suggested here: https://github.com/ghdl/ghdl/issues/2862. So these two bugs are probably related.

Paebbels, 2025-04-25:
Ubuntu 24.04 and 24.10 should give you a clean 4.1.0 version. No 5.0.1 or so.
Ubuntu 25.04 will contain 5.0.1.

The version string `5.0.1 (4.1.0.r602.g37ad91899)` shows the build of 5.0.1 but not (locally?) tagged at 5.0.1, because it's 602 commits since 4.1.0. `g` refers to Git and `37ad91899` is the SHA hash.
`37ad91899` is where tag `v5.0.1` points to.

Are you sure, you use a Ubuntu stable/LTS 24.04?

miklos-abdul, 2025-04-28:
Yes, I do use Ubuntu 24.04, which did give me clean 4.1.0. I then installed 5.0.1 (4.1.0.r602.g37ad91899) from the git release in case the bug was fixed in a newer version (it wasn't).

tgingold, 2025-05-01:
I am not sure how to reproduce the issue.  What is the `makefile.py` file ?