ghdl/ghdl #2983: Wrong version tag prefix

Remi4001, 2025-04-26:
**Description**
Not a bug with ghdl, but i think there is a typo in the latest version tag `b6.0.0`. 

**Expected behaviour**
It should be `v6.0.0` (unless there is a good reason for the `b`), because for packaging (at least in `python-pyghdl-git` in the AUR), `git describe --long` uses the version tag and then only the leading v is stripped leading to wrong version comparisons (b6.0.0 < 5.0.0) and attempts to update.

**Context**
- OS: Linux (Arch)
- Origin:
  - [x] Package manager: `yay` (`python-pyghdl-git` in AUR)
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `4c416ec3682f8caf391b5d87345a89e20e8e9344`

**Additional context**
This should be low priority. The only problem is that `yay` tries to update ghdl when it shouldn't and i have to ignore it.

Paebbels, 2025-04-26:
@tgingold maybe it's best to release a `5.1.0` directly on `main` branch to fix it and clean it up.

Remi4001, 2025-07-12:
The tag seems to have been deleted since then, so I'll just close this issue.

closed 2025-07-12 by Remi4001

Paebbels, 2025-07-14:
@Remi4001 by releasing v5.1.0, the workaround `b6.0.0` was needed anymore. The count "commits since last tag" should be correct and count towards `v5.1.0`.