ghdl/ghdl #2988: AttributeError: property 'message' of 'LibGHDLException' object has no setter

ptracton, 2025-05-01:
**Context**
I cloned the git repository on April 29th, 2025.  Git hash 6ed7f58e40641c36f1765b436a1ccb5cf26c8af2
This is on Linux Mint 22.1.  Clang version 18.1.3.  g++ version 13.3.0.  Python version 3.12.3.
I was able to build and intall GHDL and GHDL-LS.

**Description**
Any run of ghdl-ls causes an AttributeError.  

$ ghdl-ls
Traceback (most recent call last):
  File "/home/ptracton/CSUN/CSUN_ECE_VHDL/Common/Python/CSUN_PYTHON/bin/ghdl-ls", line 5, in <module>
    from pyGHDL.cli.lsp import main
  File "/home/ptracton/CSUN/CSUN_ECE_VHDL/Common/Python/CSUN_PYTHON/lib/python3.12/site-packages/pyGHDL/cli/lsp.py", line 54, in <module>
    import pyGHDL.libghdl as libghdl
  File "/home/ptracton/CSUN/CSUN_ECE_VHDL/Common/Python/CSUN_PYTHON/lib/python3.12/site-packages/pyGHDL/libghdl/__init__.py", line 206, in <module>
    libghdl = _initialize()
              ^^^^^^^^^^^^^
  File "/home/ptracton/CSUN/CSUN_ECE_VHDL/Common/Python/CSUN_PYTHON/lib/python3.12/site-packages/pyGHDL/libghdl/__init__.py", line 187, in _initialize
    _libghdl_path = _get_libghdl_path()
                    ^^^^^^^^^^^^^^^^^^^
  File "/home/ptracton/CSUN/CSUN_ECE_VHDL/Common/Python/CSUN_PYTHON/lib/python3.12/site-packages/pyGHDL/libghdl/__init__.py", line 177, in _get_libghdl_path
    ex = LibGHDLException(f"Cannot find pyGHDL shared library '{libGHDLSharedLibraryFile}'.")
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ptracton/CSUN/CSUN_ECE_VHDL/Common/Python/CSUN_PYTHON/lib/python3.12/site-packages/pyGHDL/libghdl/__init__.py", line 68, in __init__
    super().__init__(message)
  File "/home/ptracton/CSUN/CSUN_ECE_VHDL/Common/Python/CSUN_PYTHON/lib/python3.12/site-packages/pyGHDL/__init__.py", line 77, in __init__
    self.message = message
    ^^^^^^^^^^^^
AttributeError: property 'message' of 'LibGHDLException' object has no setter

**Expected behaviour**
I expect ghdl-ls to not have an AttributeError.  I am planning on using it in emacs for VHDL LSP support 

**How to reproduce?**
See above text, just run ghdl-ls from the command line.

tgingold, 2025-05-01:
I agree the python code looks incorrect, but was pyGHDL correctly installed ?  The shared library cannot be found.

Paebbels, 2025-05-01:
I also noticed a week ago, that under certain conditions, this message is shown. I debugged it and it's already in my set of fixed for the next push to GHDL. The problem is in the exception class itself. So when an exception is raised, it can't raise it due to an exception (AttributeError) while raising it.

Meanwhile, the exception it wanted to raise for you is: `Cannot find pyGHDL shared library '{libGHDLSharedLibraryFile}'.` So independent of this internal error, your setup is not correct and libghdl...so/dll isn't found.

How did you install ghdl-ls?  
Did you build a Python package and the installed the wheel via pip?

GHDL offers nightly builds including latest wheels. These come with ghdl-ls.
See https://github.com/ghdl/ghdl/releases/tag/nightly

ptracton, 2025-05-01:
I installed ghdl by ./configure; make; make install.
I installed ghdl-ls by "pip install ." from the ghdl/ directory that I cloned the repository into.

Paebbels, 2025-05-01:
@tgingold should we offer a make target to properly install libghdl.so and pyGHDL?

`make pyGHDL`
  * build libghdl if not already done by `make`
  * copy libghdl...so into `pyGHDL/lib/`
  * call `python setup.py bdist_wheel`

`make install_pyGHDL` 
* call `pip install dist/pyghdl*.whl`

We could also have target `make` call `install_pyGHDL`, then it's always fully installed also als root

@ptracton:  
1. For properly installing the Python code, please copy libghdl....so into `pyGHDL/lib`.
2. Run `python setup.py bdist_wheel`
3. Run `sudo pip install -U dist/pyghdl*.whl`

Paebbels, 2025-05-01:
Addendum:

You can also copy the so file to `/home/ptracton/CSUN/CSUN_ECE_VHDL/Common/Python/CSUN_PYTHON/lib/python3.12/site-packages/pyGHDL/lib/`

ptracton, 2025-05-01:
I did the following succesfully:
- build libghdl if not already done by make
- copy libghdl...so into pyGHDL/lib/
- call python setup.py bdist_wheel

These steps failed:
- make install_pyGHDL  : make: *** No rule to make target 'install_pyGHDL'.  Stop
- pip install dist/pyghdl*.whl : error: externally-managed-environment

I then copied the libghdl.so to  /home/ptracton/CSUN/CSUN_ECE_VHDL/Common/Python/CSUN_PYTHON/lib/python3.12/site-packages/pyGHDL/lib/

I then ran ghdl-ls and it did not crash.
I ran emacs and it initialized and started LSP mode for VHDL with out crashing.  

It looks like this libghdl.so is not getting copied into the correct location during installation.  
Is this due to the fact that I working from a virtualenv python environment and not the base OS version of python?

commit 19b02080 (2025-05-09): pyGHDL/__init__.py: avoid AttributeError

Fix #2988
closed 2025-05-09 by tgingold