ghdl/ghdl #2995: Synthesis generates invalid verilog - constants are split among generate statements

marzoul, 2025-05-12:
**Description**
GHDL generates invalid verilog.
More precisely : there are non-constant signals used in initial statements.

**Expected behaviour**
Generate valid verilog.

**How to reproduce?**

```vhd :file: top.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity top is
	generic (
		WDATA     : natural := 8;
		BLOCKS_NB : natural := 2
	);
	port (
		clk        : in  std_logic;
		write_en   : in  std_logic_vector(BLOCKS_NB-1 downto 0);
		write_data : in  std_logic_vector(WDATA-1 downto 0);
		read_data  : out std_logic_vector(BLOCKS_NB*WDATA-1 downto 0)
	);
end top;

architecture synth of top is

	signal reg_blocks : std_logic_vector(BLOCKS_NB*WDATA-1 downto 0) := (others => '0');

begin

	gen_blocks : for h in 0 to BLOCKS_NB-1 generate

		process(clk)
		begin
			if rising_edge(clk) then

				if write_en(h) = '1' then
					reg_blocks((h+1)*WDATA-1 downto h*WDATA) <= write_data;
				end if;

			end if;
		end process;

	end generate;

	read_data <= reg_blocks;

end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a top.vhd
ghdl synth --out=verilog -o=top.v top
```

The generated verilog is the following :
```verilog
module top
  (input  clk,
   input  [1:0] write_en,
   input  [7:0] write_data,
   output [15:0] read_data);
  localparam [15:0] n1 = 16'b0000000000000000;
  reg [15:0] reg_blocks;
  wire n5;
  wire n14;
  wire [7:0] n20;
  wire [7:0] n21;
  wire [7:0] n22;
  reg [7:0] n23;
  wire [7:0] n24;
  wire [7:0] n25;
  wire [7:0] n26;
  reg [7:0] n27;
  wire [15:0] n28;
  assign read_data = reg_blocks; //(module output)
  /* top.vhd:21:16  */
  always @*
    reg_blocks = n28; // (isignal)
  initial
    reg_blocks = 16'b0000000000000000;
  /* top.vhd:31:44  */
  assign n5 = write_en[0]; // extract
  /* top.vhd:31:44  */
  assign n14 = write_en[1]; // extract
  assign n20 = n1[15:8]; // extract              // Should be const
  assign n21 = reg_blocks[15:8]; // extract
  /* top.vhd:29:25  */
  assign n22 = n14 ? write_data : n21;
  /* top.vhd:29:25  */
  always @(posedge clk)
    n23 <= n22;
  initial
    n23 = n20;                                           // Error triggered here
  /* top.vhd:29:25  */
  assign n24 = n1[7:0]; // extract              // Should be const
  assign n25 = reg_blocks[7:0]; // extract
  /* top.vhd:29:25  */
  assign n26 = n5 ? write_data : n25;
  /* top.vhd:29:25  */
  always @(posedge clk)
    n27 <= n26;
  initial
    n27 = n24;                                          // Error triggered here
  /* top.vhd:29:25  */
  assign n28 = {n23, n27};
endmodule
```
Note : The internal Yosys verilog parser does not generate the error, but the Slang plugin does.
Respective command lines :
```
# This reports no error, looks too permissive
yosys -l yosys.log -p "synth_xilinx -arch xc7 -top top ; write_json top.json" top.v
# This says error
yosys -l yosys.log -m slang -p "read_slang -top top top.v ; synth_xilinx -arch xc7 -top top ; write_json top.json"
```

**Context**
Please, provide the following information:

- OS: Archlinux
- Origin: Built from sources: `commit 1061f0cb17e01a3b819b1d80a06c705ec5116672` from 2025-05-09

closed 2025-05-12 by tgingold

marzoul, 2025-05-12:
Thanks !
I observe the following diff on the produced verilog :
```
37c37
<     n23 = n20;
---
>     n23 = n1[15:8];
47c47
<     n27 = n24;
---
>     n27 = n1[7:0];
```

The wires n20 and n24 are now unused, but still declared and assigned.
Just saying in case this is undesired indeed.

tgingold, 2025-05-13:
Yes, it is the same for the vhdl output.  This could be improved, but for later.

commit 62c0be69 (2025-05-12): netlists-disp_verilog: display constant expressions in initial

Fix #2995