ghdl/ghdl #3000: TYPES.INTERNAL_ERROR : netlists-disp_verilog.adb:1336 when synthesising to Verilog

aquohn, 2025-05-19:
**Description**
When synthesising a relatively large design (~2.6kLOC) to Verilog, I encounter the above-mentioned error. However, synthesising it to the other backends (`vhdl`, `raw-vhdl`, `dump`, `raw`) succeeds.

After trying to individually synthesise the modules in the design, I have found that two separate submodules (call them A and B), which use some shared components but do not include each other, trigger the error. The written-out Verilog stops abruptly while writing out the signal declarations for the module, and both have fairly long signal lists (>320 signals).  However, the top-level module throws the error when writing the signal declaration for a submodule which is _not_ either A or B.

All modules (top, A and B) use multiple generic submodules; I'm not sure if that's part of the problem.

Running the output of the `raw-vhdl` synthesis through `ghdl` to synthesise it to Verilog also fails with the same error and the similar output (abrupt termination partway through the signal declarations).

**How to reproduce?**

I have not been able to isolate an MWE that I am able to share, but would appreciate any advice on debugging the failing code myself.

**Context**
- OS: openSUSE Tumbleweed
- Origin: oss-cad-suite

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 6.0.0-dev (b6.0.0.r125.g56a4d3e4a.dirty) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
<directory omitted>
Command line:
ghdl synth -v --out=verilog module_top.vhd -e module_top
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : netlists-disp_verilog.adb:1336
******************************************************************

tgingold, 2025-05-20:
You are not using the latest build of ghdl, so maybe this issue has been fixed (in particular, there have been changes in the verilog output).
Otherwise, I need a reproducer to investigate.  That could be the synthesized output in vhdl and if needed you can deplete it.

aquohn, 2025-05-21:
I've tried with the latest Docker image, ghdl:5.0.0-rc.1-gcc-ubuntu-24.04, and ghdl/ghdl:5.0.1-gcc-ubuntu-24.04, but all exhibit the same error. I forgot to mention that I have a copy of the successfully generated output from when I last ran the build process, but I'm not sure which version of ghdl was used to build it (although the mtime indicates it was 14 January 2025). The earlier docker images also produced these call stacks though, which may be helpful:

5.0.0:
```
0x456907 0x455f70 0x674067 0x6754c3 0x6539be 0x40ba27 0x7f257afaa1c8 0x7f257afaa289 0x40ba83 0xfffffffffffffffe
```

5.0.1:
```
0x456907 0x455f70 0x674057 0x6754b3 0x6539be 0x40ba27 0x7fd6c0a741c8 0x7fd6c0a74289 0x40ba83 0xfffffffffffffffe
```

I'll continue trying to hack the code down to an MWE that I can post. Thanks!

aquohn, 2025-05-21:
I've found a public repository where this error is triggered: https://github.com/hadipourh/AES-VHDL

Clone the repository and run
```
ghdl synth -v --out=verilog --latches AES-DEC/RTL/*.vhd -e aes_dec > aes_dec.v
```
from its root. The symptoms appear similar to the issue I'm encountering.

Also, I realised that I mixed up the versions; in the latest GHDL the error is on line 1226. I will update the issue title accordingly.

commit becdaf6f (2025-05-21): netlists: clear location in cleanup before remove

Do not check location of nop.

Fix #3000
closed 2025-05-21 by tgingold