ghdl/ghdl #3002: Large constant vector init from string concat has quadratic RAM usage

marzoul, 2025-05-20:
**Description**
GHDL analysis of declaration of a large constant std_logic_vector uses extreme amounts of RAM memory.
This happens with "concat" coding style : `:= 00" & "00" & "00" & ...`
This does not happen "list" coding style : `:= ('0', '0', '0', '0', '0', '0', ...)`

**Expected behaviour**
RAM usage should increase linearly (as much as possible).

**How to reproduce?**
The reproducers are too large to be pasted here.
An archive is available at this public link (22kB) :
https://cloud.univ-grenoble-alpes.fr/s/CLdoC76XrLsLNPp

A Makefile is provided to launch a series of 3 tests, designated by their size 16k, 32, 64k.
These correspond to the number of 4-bit chunks to be concatenated.

Type `make tests-concat` to reproduce.
Type `make tests-list` for a reference with list-style implementation, which does not exhibit the issue.

It may be necessary (I don't know) to increase stack size with `ulimit -s 200000`

Resident RAM usage with list implementation: 14 MB, 22 MB, 38 MB
Resident RAM usage with concat implementation: 32 MB, 94 MB, 565 MB

**Context**
Please, provide the following information:

- OS: Archlinux
- Origin: Built from sources, fresh compilation on 2025-05-20

Paebbels, 2025-05-20:
Is this synthesis related or in general?

marzoul, 2025-05-21:
It is general because it happens right at vhdl analysis.
It is blocking for both simulation and synthesis.

The code is generated so at least the list-style workaround exists.
But the concatenated strings style is convenient, because the vhdl files are smaller, and the contents are much more convenient to manually check than the bit-banged list style.

closed 2025-05-21 by tgingold

tgingold, 2025-05-22:
Yes, but I doubt large and deep trees are created by other operators.