ghdl/ghdl #3009: ghdl compilation flag to error features of vhdl2008 that are most annoying to vhdl1993 programmers...

wm2015email, 2025-06-01:
**Is your feature request related to a problem? Please describe.**  

I’m generally happy with the improvements in VHDL-2008, but the addition of unconstrained entity‐port declarations causes significant pain for many teams (especially those with large, hierarchical package structures). In particular:

- **Entities cannot be analyzed/compiled in isolation.** As soon as an entity declares an unconstrained port, tools like GHDL refuse to analyze it until you provide a full instantiation (or a generic map) that fixes the bounds. In large designs—where package files already carry dozens of dependent types—this quickly becomes a huge time sink whenever someone adds or modifies a top-level interface.

- **Legacy workflows break.** Many older VHDL users prefer to keep entity interfaces fully constrained so that you can compile/flake-check them without wiring up every testbench or wrapper. In practice, some teams have even refused to migrate to VHDL-2008 specifically because unconstrained ports force them into extra “plumbing” before they can do a simple syntax check or lint.

- **Manual workarounds are tedious.** Developers resort to writing small dummy wrappers or adding verbose pragma annotations just to satisfy the compiler, which defeats the original goal of having flexible data widths.

I’m always frustrated when I want to run a quick “compile only” cycle on a new entity—only to discover that because I used an unconstrained port, I have to write a wrapper or add placeholder bounds before GHDL will process my file.

---

## Describe the solution you’d like  

I would like GHDL to support a **linting-only warning** whenever an entity declares unconstrained ports under VHDL-2008. Ideally, this would be controlled by a new command‐line flag (or style directive) so that teams can opt into a strict warning without turning off all VHDL-2008 support. For example:

```
ghdl -a mydesign2008.vhd --warn-unconstrained-ports
```

When this flag is enabled, GHDL would still accept unconstrained ports for simulation/analysis, but it would produce a warning like:

```
ghdl: WARNING: mydesign2008.vhd:10:1: entity “mydesign2008” uses unconstrained port “data_bus”
```

In other words:

1. **Analysis proceeds as normal.** GHDL still fully supports unconstrained types under VHDL-2008, so existing code continues to compile and elaborate exactly as before.

2. **Optional warning.** If `--warn-unconstrained-ports` is set, emit a warning for each unconstrained port (both scalar unconstrained types, unconstrained arrays, and access types with unconstrained bounds).

3. **Configurable linter behavior.** In a future release, this could be extended to treat those warnings as errors (e.g. `--error-unconstrained-ports`) or to whitelist certain entities via attributes or pragmas.

This flag would let teams enforce a “no unconstrained ports” policy during continuous integration or nightly regression runs without impacting anyone who actually wants to use that VHDL-2008 feature.

---

## Describe alternatives you’ve considered  

1. **External linting tools (e.g., ghdl-lint or vhdllint).**  
   - We’ve experimented with open-source linters that parse VHDL and report unconstrained ports. However, those tools are often slower or less up-to-date with GHDL’s VHDL-2008 parser. They also require an extra step in the build pipeline, and they don’t integrate as cleanly with GHDL’s error reporting (line numbers, context, etc.).

2. **Manual code review and coding guidelines.**  
   - Some teams simply document “avoid unconstrained ports” in their style guide. In practice, though, code reviewers miss things, and it’s easy for someone to slip in an unconstrained declaration. Without compiler support, there’s no foolproof way to catch this at compile time.

3. **Use constrained ranges everywhere.**  
   - It’s always possible to replace `std_logic_vector` with a subtype that has explicit bounds, or use unconstrained generic constants. But that defeats the purpose of writing a reusable, parameterized interface. Ultimately, it becomes a contest between flexibility (unconstrained ports) and maintainability (fully constrained).

Because none of these alternatives address the core pain point—“I want to compile entities in isolation, but I don’t want unconstrained ports ever to slip by”—having a built-in GHDL warning feels like the most maintainable, zero-overhead solution.

---

## Additional context  
- **VHDL-2008 unconstrained ports** allow you to write things like:  
  ```vhdl
  entity fifo is
    port (
      clk     : in  std_logic;
      rst     : in  std_logic;
      data_in : in  std_logic_vector;      -- unconstrained
      data_out: out std_logic_vector       -- unconstrained
    );
  end entity;
  ```

  This is very convenient (you don’t have to repeat a length on every use), but under the hood, GHDL must wait until you fully bind `data_in`/`data_out` in an instantiation before it knows the range. In contrast, a constrained declaration would look like:  
  ```vhdl
  entity fifo is
    generic ( WIDTH : positive );
    port (
      clk      : in  std_logic;
      rst      : in  std_logic;
      data_in  : in  std_logic_vector(WIDTH-1 downto 0);
      data_out : out std_logic_vector(WIDTH-1 downto 0)
    );
  end entity;
  ```

  With the generic style, GHDL can check everything immediately.

- **Industry impact.**  
  • Some aerospace/defense shops still compile VHDL with vendor tools that do not fully support VHDL-2008. They find that an unconstrained port in a top-level entity breaks their golden regression suite. Having a GHDL-native warning allows them to catch noncompliant code before pushing to the vendor toolchain.  
  • Educational code still often uses older VHDL. In university labs, instructors prefer students to declare bit widths explicitly. A warning flag makes it easy to teach the “right” style without disabling VHDL-2008 altogether.

- **Possible future enhancements.**  
  - Beyond a simple warning, GHDL could provide a `--fix-unconstrained-ports` mode that automatically generates stubs or constrained wrappers when analyzing an entity.  
  - Integrate with CI: teams could set “treat warnings as errors” so that any unconstrained port blocks the build.

Thank you for considering this request. A lint‐only warning for unconstrained ports would let teams gradually adopt VHDL-2008 without breaking existing build processes or slowing down development.

tgingold, 2025-06-01:
Please, provide a testcase (a simple input file and the commands you entered).
It is definitely possible to analyze an entity with unconstrained ports.

However, you cannot simulate an entity with unconstrained ports - but that's not new.

I am a little bit surprised with this issue: in a sense, there is nothing new in vhdl 2008 with this issue: even with vhdl 93, it was possible to have unconstrained ports.  So there must be a misunderstanding.

wm2015email, 2025-06-02:
Thank you for considering this feature request…

That’s a good point that it lets you compile unconstrained entity ports in
vhdl-93 but won’t let you simulate them…. at least with vhdl-1993 this was
an error at some point in the compilation chain and people would remove the
unconstrained entity ports because they can’t simulate their designs…. the
problem is that vhdl-2008 doesn’t have this message… and I wanted to keep
everything with constrained entity ports as a project style… with out a
linting flag to print a warning for this it always creeps into the
vhdl-2008 code again and ruins the entire
project again… then people start demanding that nothing be written in
vhdl-2008 because this is a terrible feature that ruins code and thus we
can only use vhdl1993

 I just wanted a linting flag to enforce this style of coding in vhdl 2008
… a linting flag would be very easy to embed it in  a ghdl compilation
script for the project and then they we would just remove the unconstrained
entity ports because they are not allowed to have warning in their
compilation to check in their vhdl code..

tgingold, 2025-06-02:
> the problem is that vhdl-2008 doesn’t have this message

That's what I am challenging.  Do you have an example of such an entity ?

aspdigital, 2025-09-08:
> Thank you for considering this feature request…
>
> That’s a good point that it lets you compile unconstrained entity ports in
> vhdl-93 but won’t let you simulate them…. at least with vhdl-1993 this was [...]

 Let's assume that your DUT has unconstrained ports.

I think you do not understand how to write test benches. You do not simulate the DUT directly, which is what I think you are trying to do. A proper test bench _instantiates_ the DUT, and the actuals on the DUT ports must be constrained. 

As Tristan points out, this is nothing new.