ghdl/ghdl #3011: Support -gGENERIC=VALUE for generics of type std_logic

goustandp, 2025-06-02:
**Description**
Setting a VHDL generic values using the ghdl -r -gGENERIC=VALUE failed returning the following error 

```
ghdl:error: override for generic "G_RST" is out of bounds
```

It seems that std_logic is not supported. Using a 1 bit lenght std_logic_vector can be used as a work around.

**How to reproduce?**
 For example, the following entity will raise an error:

```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

entity toto is
  generic(
    G_RST : std_logic
  );
  port(CLK : in std_logic;
                RST : in std_logic;
                A : out std_logic
  );
 end entity;

architecture rtl of toto is

begin

process(CLK)
begin
  if rising_edge(CLK) then
                if (RST = G_RST) then
                        A <= '0';
                else
                        A <= '1';
                end if;
        end if;
end process;

end rtl;
```

```
ghdl -r -gG_RST=1 toto
ghdl:error: override for generic "G_RST" is out of bounds
```

But changing G_RST type to std_logic_vector is OK :
```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

entity toto is
  generic(
    G_RST : std_logic_vector(0 downto 0)
  );
  port(CLK : in std_logic;
                RST : in std_logic;
                A : out std_logic
  );
 end entity;

architecture rtl of toto is

begin

process(CLK)
begin
  if rising_edge(CLK) then
                if (RST = G_RST(0)) then
                        A <= '0';
                else
                        A <= '1';
                end if;
        end if;
end process;

end rtl;

```

**Context**

- OS: Ubuntu 20.04 (using WSL2)
- Origin:
  - GHDL 6.0.0-dev (b6.0.0.r159.gdadb10f98) [Dunoon edition]
  - Compiled with GNAT Version: 13.3.0
  - static elaboration, mcode JIT code generator

tgingold, 2025-06-02:
Did you try `-gg_RST=\'1\'` ?

goustandp, 2025-06-03:
Thank for response. Using backslashes works :

``` 
-gG_RST=\'0\'
```
or 
```
-gG_RST=\'1\'
```

tgingold, 2025-06-06:
The reason is that an enumeration can both have character literals (like '0', or 'A') and simple name (like CR, LF).
We need a way to distinguish 'A' and A.  But maybe the interface can be more user friendly.

closed 2025-06-06 by tgingold