ghdl/ghdl #3028: unhandled (static) function: IIR_PREDEFINED_IEEE_MATH_REAL_POW_INT_REAL

MartinBober, 2025-07-07:
**Description**
The IEEE real power functions seems to not be implemented.

**Expected behaviour**
Code below should compile

**How to reproduce?**

```vhd :file: ent.vhd

library ieee;
use ieee.std_logic_1164.all;
use ieee.math_real.all;

entity ent is
end entity;

architecture a of ent is
    constant NO_PORTS_C    : integer := 8;
    constant NO_MUXPORTS_C : integer := 2;
    constant NEEDS_POW : integer :=
        integer(
            ceil(
                real(NO_PORTS_C) /                -- 8.0
                ( NO_MUXPORTS_C ** real(1) )      -- 2 ** 1.0  <-- crash here
            )
        );
begin
    assert NEEDS_POW > 0
      report "NEEDS_POW = " & integer'image(NEEDS_POW);
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

**Context**

- OS: x86_64-w64-mingw32
- Origin: `https://github.com/ghdl/ghdl/releases/download/v5.0.1/ghdl-mcode-5.0.1-mingw64.zip`

Paebbels, 2025-07-07:
Please provide the error message of GHDL.

MartinBober, 2025-07-07:
Sorry for not including that in the first place.

```
.\poc_math_real_po_int_real.vhd:15:33:error: unhandled (static) function: IIR_PREDEFINED_IEEE_MATH_REAL_POW_INT_REAL
                ( NO_MUXPORTS_C ** real(1) )      -- 2 ** 1.0  <-- crash here
```

closed 2025-07-08 by tgingold