ghdl/ghdl #3044: Unhandled predefined IEEE operator (with ieee.numeric_bit only)

patrick-studer, 2025-09-01:
**Description**
I get some simulation run-time errors related to **"unhandled predefined IEEE operator"**.
For me, the error message looks like the `ieee.numeric_bit` package is wrongly parsed. The message truncates the last character of the operator.

```
test_pkg.vhd:29:31:error: unhandled predefined IEEE operator """
        temp3 := resize(temp2 + 2, nx);
                              ^
../../src/ieee/v93/numeric_bit.vhdl:93:12:error:  declared here
  function "+" (L: UNSIGNED; R: NATURAL) return UNSIGNED;
           ^
test_pkg.vhd:30:31:error: unhandled predefined IEEE operator """
        temp4 := resize(temp3 - 1, nx);
                              ^
../../src/ieee/v93/numeric_bit.vhdl:127:12:error:  declared here
  function "-" (L: UNSIGNED; R: NATURAL) return UNSIGNED;
           ^
test_pkg.vhd:31:31:error: unhandled predefined IEEE operator """
        temp5 := resize(temp4 * 4, nx);
                              ^
../../src/ieee/v93/numeric_bit.vhdl:161:12:error:  declared here
  function "*" (L: UNSIGNED; R: NATURAL) return UNSIGNED;
           ^
test_pkg.vhd:32:31:error: unhandled predefined IEEE operator """
        temp6 := resize(temp5 / 2, nx);
                              ^
../../src/ieee/v93/numeric_bit.vhdl:204:12:error:  declared here
  function "/" (L: UNSIGNED; R: NATURAL) return UNSIGNED;
           ^
test_pkg.vhd:33:31:error: unhandled predefined IEEE operator ""mo"
        temp7 := resize(temp6 mod 15, nx);
                              ^
../../src/ieee/v93/numeric_bit.vhdl:284:12:error:  declared here
  function "mod" (L: UNSIGNED; R: NATURAL) return UNSIGNED;
           ^
test_pkg.vhd:34:31:error: unhandled predefined IEEE operator ""re"
        temp8 := resize(temp7 rem 11, nx);
                              ^
../../src/ieee/v93/numeric_bit.vhdl:243:12:error:  declared here
  function "rem" (L: UNSIGNED; R: NATURAL) return UNSIGNED;
           ^
```

- The error only occurs with numeric_bit (not with numeric_std)
- The error only occurs, when code is called in the architecture definition (e.g. to define a constant)

**Expected behaviour**
No error, even numeric_bit is used in package functions and called in the architecture definition region (before begin keyword).

**How to reproduce?**

Use test files provided here: [test.zip](https://github.com/user-attachments/files/22077435/test.zip)

1. Run test_tb.vhd to see the error:
```
 ghdl -a test_pkg.vhd
 ghdl -a test_tb.vhd
 ghdl -r test_tb
```

2. Comment out constant definition in the architecture (line 16+17 or test_tb.vhd)
  - Now it will work.

3. Change both files to use numeric_std instead of numeric_bit library
  - It works in both cases - with and without constant definitions...

**Context**
Please, provide the following information:

- OS: Windows 10 + Linux
- Origin:
  - [x] Package manager: `5.1.1` (GHDL - v5.1.1 - Windows (x86-64, MinGW64, standalone) - mcode backend)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2025-09-02:
I am fixing this.  There is special code to handle numeric_bit, but as it is not widely used, many operations are not yet supported (during elaboration).

patrick-studer, 2025-09-02:
Thank you very much for the fast reply.