ghdl/ghdl #3050: bad elaboration order of objects

Elkniwt, 2025-09-09:
**Description**
I get the following, given a rather extensive VHDL code base:

```
ghdl:error: synth: bad elaboration order of objects

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 6.0.0-dev (5.1.1.r60.g8c69dbd51) [Dunoon edition]
Compiled with GNAT Version: 13.3.0
Target: x86_64-linux-gnu
/home/jspeter/work/src/git/vhdl-ws-xv/.xv-aa-shared/src/internal/sim_tests/shared/ws_testers_package/ws_adc_and_dac_tester/ghdl.run/
Command line:
ghdl -e --ieee=synopsys -fexplicit --warn-no-binding ws_sim_top
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : elab-vhdl_context.adb:321
******************************************************************
    while executing
"exec ghdl -e --ieee=synopsys -fexplicit --warn-no-binding $::SIM_TOP"
```

**Expected behaviour**
I would expect elaboration to succeed normally... or at least some debug info pointing to what the problem may be.  I commented out the check that emits the "bad elaboration" message in the codebase and then got an error in Create_Subtype_Object.  I know that a new subtype had been added to our codebase that wasn't strictly necessary, so I reworked it to avoid that subtype, yet the error persists.  Is there anything I can print, near the time of error, that may help debug this?

**How to reproduce?**
I'm not sure I'm allowed to share this code.  I notice that I can comment out one component and the issue goes away.  If I comment out that component's architecture contents, the issue remains.

**Context**
Please, provide the following information:

- OS: Ubuntu 24.04.3 LTS
- Origin: (I've tried two versions)
  - [X] Package manager: `version` GHDL release: 4.1.0 (Ubuntu 4.1.0+dfsg-0ubuntu2.1) [Dunoon edition]
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `commit SHA` commit 8c69dbd517970e7fc2e86ae3d6541c37d49bbcef

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 6.0.0-dev (5.1.1.r60.g8c69dbd51) [Dunoon edition]
Compiled with GNAT Version: 13.3.0
Target: x86_64-linux-gnu
/home/jspeter/work/src/git/vhdl-ws-xv/.xv-aa-shared/src/internal/sim_tests/shared/ws_testers_package/ws_adc_and_dac_tester/ghdl.run/
Command line:
ghdl -e --ieee=synopsys -fexplicit --warn-no-binding ws_sim_top
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : elab-vhdl_context.adb:321
******************************************************************
```

**Additional context**
I have the git repo, and am happy to print out any relevant information, if needed.  I'm not too familiar with Ada, but have done Pascal and VHDL programming.

tgingold, 2025-09-09:
Well, you can try to debug by yourself, but elaboration issues are usually a delicate problem.  However with a debugger you can try to see which construct is triggering the issue.

But the best way to make progress is to have a reproducer.  If the instantiated components can be empty, you aren't far from a small reproducer.

Elkniwt, 2025-09-10:
Perfect!  Let me know if you want me to test the fix.  I have a few other instances of similar generics (though they're mostly the same).

Elkniwt, 2025-09-10:
Works great.  Thank you.