ghdl/ghdl #3058: An assignment to an output port by a procedure gives the error message "port 'b' is not a formal parameter"

matthiasschweikart, 2025-09-19:
**Description**
An assignment to an output port done inside a procedure (which is declared in the architecture), without defining the port as a parameter of the procedure, creates the error message "port 'b' is not a formal parameter" when analyzing the VHDL (with and without using "--std=08").

**Expected behaviour**
In my opinion this is legal VHDL and should not produce any error message.

**How to reproduce?**
The issue can be reproduced by:

```vhd :file: demo.vhd
library ieee;
use ieee.std_logic_1164.all;
entity demo is
  port (
    b : out std_logic
  );
end entity;
architecture rtl of demo is
  procedure drive_b is
  begin
    b <= '1';
  end procedure;
begin
  process
  begin
    drive_b;
    wait;
  end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a demo.vhd
```

**Context**
GHDL-Version:
GHDL 5.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 14.2.0
 llvm 19.1.4 code generator

- OS: Win 11
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

diogratia, 2025-09-19:
I use an aliased find invocation to find partial error strings in the ghdl source tree:

```none
%: alias adafind
find . -iname \*.ad\[bs\] -exec grep -H !* \{\} \;
```
(This a csh alias)

It reports

```none
%: adafind -i "is not a formal parameter"
./vhdl/vhdl-sem_stmts.adb:            Error_Msg_Sem (+Stmt, "%n is not a formal parameter", +Sig_Object);
```

which tells the filename and with an additional grep flag the line number which can vary depending on the commit level of the source tree.

finding the partial string in the vhdl-sem_stmts.adb source file reveals the LRM subsection number, here 8.4.1 which is -1993, -2008 would be stated explicitly:

```ada
   procedure Sem_Add_Driver (Sig : Iir; Stmt : Iir)
   is
      Sig_Object : Iir;
      Sig_Object_Type : Iir;
   begin
      if Sig = Null_Iir then
         return;
      end if;
      Sig_Object := Get_Object_Prefix (Sig);
      Sig_Object_Type := Get_Type (Sig_Object);

      --  LRM 4.3.1.2 Signal Declaration
      --  It is an error if, after the elaboration of a description, a
      --  signal has multiple sources and it is not a resolved signal.

      --  Check for multiple driver for a unresolved signal declaration.
      --  Do this only if the object is a non-composite signal declaration.
      --  NOTE: THIS IS DISABLED, since the assignment may be within a
      --  generate statement.
      if False
        and then Get_Kind (Sig_Object) = Iir_Kind_Signal_Declaration
        and then Get_Kind (Sig_Object_Type)
        not in Iir_Kinds_Composite_Type_Definition
        and then not Get_Resolved_Flag (Sig_Object_Type)
      then
         if Get_Signal_Driver (Sig_Object) /= Null_Iir and then
           Get_Signal_Driver (Sig_Object) /= Current_Concurrent_Statement
         then
            Error_Msg_Sem (+Stmt, "unresolved %n has already a driver at %l",
                           (+Sig_Object, +Get_Signal_Driver (Sig_Object)));
         else
            Set_Signal_Driver (Sig_Object, Current_Concurrent_Statement);
         end if;
      end if;

      --  LRM 8.4.1
      --  If a given procedure is declared by a declarative item that is not
      --  contained within a process statement, and if a signal assignment
      --  statement appears in that procedure, then the target of the
      --  assignment statement must be a formal parameter of the given
      --  procedure or of a parent of that procedure, or an aggregate of such
      --  formal parameters.
      --  Similarly, if a given procedure is declared by a declarative item
      --  that is not contained within a process statement and if a signal is
      --  associated with an INOUT or OUT mode signal parameter in a
      --  subprogram call within that procedure, then the signal so associated
      --  must be a formal parameter of the given procedure or of a parent of
      --  that procedure.
      if Current_Concurrent_Statement = Null_Iir
        or else (Get_Kind (Current_Concurrent_Statement)
                 not in Iir_Kinds_Process_Statement)
      then
         --  Not within a process statement.
         if Current_Subprogram = Null_Iir then
            --  not within a subprogram: concurrent statement.
            return;
         end if;

         --  Within a subprogram.
         if Get_Kind (Sig_Object) = Iir_Kind_Signal_Declaration
           or else not Is_Parameter (Sig_Object)
         then
            Error_Msg_Sem (+Stmt, "%n is not a formal parameter", +Sig_Object);
         end if;
      end if;
   end Sem_Add_Driver;
```
This is found in -1993  8.4.1 Updating a projected output waveform.

Upon seeing your issue I had searched through the -2008 standard not recalling explicit language of the semantic requirement and found it in the last paragraph of -2008 10.5.2.2 Executing a simple assignment statement:

>If a given procedure is declared by a declarative item that is not contained within a process statement, and if
a simple waveform assignment statement appears in that procedure, then the target of the simple waveform
assignment shall be a formal parameter of the given procedure or of a parent of that procedure, or an
aggregate of such formal parameters. Similarly, if a given procedure is declared by a declarative item that is
not contained within a process statement, and if a signal is associated with an inout or out mode signal
parameter in a subprogram call within that procedure, then the signal so associated shall be a formal
parameter of the given procedure or of a parent of that procedure.

So the idea here is to point out there such misapprehensions occur regularly as VHDL users and implementers progress in gaining VHDL experience and can be overcome by thorough inculcation in the standard.

From the Ada procedure name you can see the semantic issue deals with identifying drivers, wherein the standard does not require peering into a procedure not defined in the process statement declarative part to identify signal drivers, instead relying on parameter declarations while ghdl suggests a fix besides moving the procedure body to the process declarative  part.

closed 2025-09-20 by matthiasschweikart

diogratia, 2025-09-21:
IEEE Std 1076-1987  
8.3.1 Updating a Projected Output Waveform  

The last paragraph and the following note:

>...  
>  
>If a given procedure is declared by a declarative item that is not contained within a process
statement, and a signal assignment statement appears in that procedure, then the target of the
assignment statement must be a formal parameter of the given procedure or of a parent of that
procedure, or an aggregate of such formal parameters.  
 >   
>*Note:*  
>
>If a right-hand side value expression is either a numeric literal, or an attribute that yields a
result of type universal_integer or universal_real, then an implicit type conversion is
performed.  
>  
>The above rules guarantee that the driver affected by a signal assignment statement is always
statically determinable if the signal assignment appears within a given process (including
the case in which it appears within a procedure that is declared within the given process). In
this case, the affected driver is the one defined by the process; otherwise, the signal assignment
must appear within a procedure, and the affected driver is the one passed to the procedure along
with a signal parameter of that procedure.

And we see the reason a little clearer from the note which is more expansive than those found in later revisions.